Frix-x/klippain #227: Modular Integration of ERCF

Surion79, 2023-06-10:
currently, if ERCF is out of order, it is rather complex/intransparent, what must be done to have it removed out of Klippain.
Currently the include in printer.cfg, the mcu.cfg must be edited, and everything has to be commented out.
I do wonder (hence this issue), if it could be integrated a bit more modular, that only a single line is required to add/remove ERCF configuration.

Frix-x, 2023-06-10:
Hi,

Can you elaborate on your issue regarding ERCF ?
@elpopo-eng, @Benoitone and another french guys are running fine using the last develop branch.

A one line include for the ERCF is indeed something that would be very good, but it's not really possible due to how HH is designed right now. But to deactivate fully the ERCF, you only need to remove the 5 lines (instead of 1) and nothing more. Everything should be deactivated if you put them back like that: https://github.com/Frix-x/klippain/blob/2aab85516013ef688c9260b95ff22cc1dfff675f/user_templates/printer.cfg#L178-L188

Surion79, 2023-06-10:
That is not sufficient. Unfortunately, all the lines in mcu.cfg must be commented out as well.

Benoitone, 2023-06-10:
Mcu is mcu hardware? So I think if you remove hardware you must delete its configuration from mcu. It's not only valid for ercf board, but also for all mcu you add or delete?

Surion79, 2023-06-12:
You need to remove the ercf includes, and then all lines in MCU linked to the ercf mcu configuration and its assignments in mcu.cfg itself. this includes everything between [ercf] (as first line) including the last line of manual assignments between mcu definitions and direct assignments. And of course any residuals left beind in overrides.cfg referring to commented blocks of ercf configuration. (if not editied in the ercf_hardware.cfg directly)

Surion79, 2023-06-12:
a possible idea would be to have some sort of other arrangement regarding the mcu.cfg.
currently everything is stuffed there and not modular at all for e.g. exchanging toolhead boards, or removing a toolhead or ercf. it would be nice to have some sort of possibility to have just one line to change to get switch a toolhead or mainboard and then just update the existing default config instead of manually transferring data. I am not sure if it is reliable/feasable enough to write a script to do some replacement of config files.
i was thinking about something similar to the printer.cfg where all includes are added and you only select the ones you have.
Just a braintormy idea :)

Surion79, 2023-06-23:
i do wonder, if it is an option to separate ercf mcu in an own file and add the include in the section of printer.cfg
like "Include the following file if you use a dedicated ercf mcu"

Frix-x, 2023-08-16:
Closing for inactivity.
Also that the new HHv2 will come it will change a lot of thing regarding this issue and almost everything should be addressed.