Frix-x/klippain #229: Publish shaper calibration scripts as separate plugin?

thisiscam, 2023-06-11:
Hi,

I wonder if you would consider publishing the shaper calibration scripts as separate klipper plugins? They look great and I think is a must have for most printers. But some users (like I) might not be able to make the full switch to klippain because our existing configs are too large to be ported easily..

Thank you

Frix-x, 2023-06-12:
Hi,
Thanks for raising an issue about that.

In fact, I designed all the calibration macros (as the InputShaper automated workflow and scripts) as detached from the Klippain code. This means that you can already install them as standalone as a lot of people are already doing. The install documentation is here: https://github.com/Frix-x/klippain/blob/main/docs/features/is_workflow.md

Surion79, 2023-06-12:
> Hi,
>
>
>
> I wonder if you would consider publishing the shaper calibration scripts as separate klipper plugins? They look great and I think is a must have for most printers. But some users (like I) [...]

If you ever plan to migrate to Klippain, you could start by open an issue for a single feature request and it gets easier over time to migrate :-)

thisiscam, 2023-06-21:
> Hi, Thanks for raising an issue about that.
>
> In fact, I designed all the calibration macros (as the InputShaper automated workflow and scripts) as detached from the Klippain code. This means that [...]

Hi thanks! Yes, I found that documentation after posting this issue. You are right that manual installation might be fine. But, users installing these manually will not be able to catch any updates you push to these scripts in this repo. I would agree that these scripts seems fairly stable that maybe a separate repo is not necessary, though.

closed 2023-06-21 by thisiscam