Frix-x/klippain #232: CB1 BIGTREETECH

jha03500, 2023-06-12:
Bonjour, 

Klippain est il ou sera il compatible avec le CB1 BIGTREETECH

Merci

Frix-x, 2023-06-13:
> E.g. graph_vibrations.py has a changelog for being independant of "pi", but it is linked in line 217 as default.

Yes but it's not used. The Pi in the script is default if no parameter is sent to the script call. The thing (that is said in the changelog) is that now the script accept a parameter for that, and this is used also in the plot_graph.sh here (-k parameter):
https://github.com/Frix-x/klippain/blob/4baf0ea6aec60c3959693b95e531a336bea9a906/scripts/plot_graphs.sh#L139

So this is not a problem anymore and also work when the use is not pi :)

The only place where it's still needed and I can't use a generic like (`~/`) is in the shell_command. But this is not really a problem as it can be overriden very easily
https://github.com/Frix-x/klippain/blob/4baf0ea6aec60c3959693b95e531a336bea9a906/scripts/shell_commands.cfg#L4-L7

Surion79, 2023-06-13:
@Frix-x i think the requester likes an answer 'en français' 😀

jha03500, 2023-06-13:
Oui le français est ma langue, mais je peux faire l’effort 
Ma question est peut on installer klippain sur un BTT Pi V1.2 ? La procédure est compliquée ? 
Merci

Surion79, 2023-06-13:
La procédure n'est pas compliquée.
BTT Klipper Pi 1.2 est lent. 
(Je parle un poco peu français)

Frix-x, 2023-06-13:
Hello,

Oups, oui j'avais pas fait attention que c'était en Français ^^
Mais du coup Klippain est une configuration spéciale pour Klipper (c'est un équivalent au fichier `printer.cfg` que tu connais déjà surement). Donc cela veut dire que si Klipper fonctionne, Klippain aussi :)

jha03500, 2023-06-13:
Merci,
Pour vos explications

Frix-x, 2023-06-15:
Globalement oui, Klippain c'est ça :)

jha03500, 2023-06-15:
Merci pour votre patience 😉

closed 2023-06-16 by Frix-x