Frix-x/klippain #240: adaptive_bed_mesh isn't working with Beacon probe

R4toLoko, 2023-06-20:
Hi all,

I must be missing something but adaptive_bed_mesh doesn't appear to be working with Beacon probe.
I got it recommend by somebody over KAMP with a Trident like and beacon probe like me so it could be user mistake.

- [ x ]  `adaptive_bed_mesh.cfg` in place
- [ x ]  `[include adaptive_bed_mesh.cfg]`
- [ x ]  `printer.cfg` < `[exclude_object]`

It doesn't work so I tried method 2 added to the slicer:

- [ x ]  `SIZE={first_layer_print_min[0]}_{first_layer_print_min[1]}_{first_layer_print_max[0]}_{first_layer_print_max[1]}`

It doesn't work either.

I can see the gcode has all the data so I am not sure what is wrong.

```
SIZE=135.1_135.1_164.9_164.9
; EXECUTABLE_BLOCK_START
EXCLUDE_OBJECT_DEFINE NAME=Voron_Design_Cube_v7.stl_id_0_copy_0 CENTER=150,150 POLYGON=[[135,135],[165,135],[165,165],[135,165],[135,135]]
```
`BED_MESH_CALIBRATE` is working just fine, no errors.

This is my `bed_mesh`:

```
#####################################################################
#   Bed Mesh
#####################################################################

[bed_mesh]
speed: 500
horizontal_move_z: 4
mesh_min: 30,30 #10, 22 #Probe location
mesh_max:  270,270 #290, 280 #Probe location
probe_count: 20,20
algorithm: bicubic
```
Thank you

Frix-x, 2023-06-20:
Hi, thanks for raising an issue!

I don't see why my adaptive mesh would not work with Beacon as it's just calling the standard bed mesh with some parameters under the hood. So it should be something misconfigured on your side.

First thing I noticed is that you added only `SIZE=xxxxx` in the custom gcode from the slicer. It should be passed in your START_PRINT macro instead as a parameter **on the same line**.

Also, can you give me your `klippy.log` to see how it is called in your config and what is happening?

R4toLoko, 2023-06-20:
@Frix-x ,

Thank you for checking this out. 
Like I mentioned, it is probably me doing noob stuff, the Trident dude who recommend it, has it running just fine.
I followed the instructions here: https://github.com/Frix-x/klippain/blob/main/docs/features/adaptive_bed_mesh.md

Actually, I am checking everything again I noticed step 5 regarding `RESPOND`, I don't have that in place so I wonder if it is falling because of that.
I have added `[respond]` to my `printer.cfg` just like` [exclude_object] `but will wait for further instructions.

It has been passed in my START_PRINT macro. Please ignore the wrap, everything is in a single line like so:
`PRINT_START HOTEND={first_layer_temperature[initial_extruder]} BED=[first_layer_bed_temperature] ZOFFSET=0.04 MATERIAL=[filament_type] SIZE=first_layer_print_min[0]}_{first_layer_print_min[1]}_{first_layer_print_max[0]}_{first_layer_print_max[1]}`

[image]

I am sending you the log attached as requested.
[klippy.log.zip](https://github.com/Frix-x/klippain/files/11800852/klippy.log.zip)

Thanks a lot.

Surion79, 2023-06-20:
Which version of Klippain do you use? Your starting print macro is not the klippain macro.

Surion79, 2023-06-20:
Klippain is a full suit to configure your whole printer. This includes flexible print start and end macros and much more. Adaptive bed mesh and the vibrations scripts are part of it, but not everything is completely modular.
If you want to use KAMP, ... :-)

Frix-x, 2023-06-20:
I don't think KAMP is the solution since it's working exactly the same way as my macro. 

Unfortunately I did not have time tonight to have a look at your klippy.log but I'm pretty sure it's a quick fix. If you don't mind I'll come back to you tomorrow morning :)

R4toLoko, 2023-06-20:
@Frix-x , 
This is my first Voron and I would prefer to don't screw it up, this LDO Trident has been a workhorse so I wanna keep its LDO `printer.cfg` as it's.

This isn't an emergency and can wait :)
Thank you for taking your time looking into this but again, it isn't an emergency.
I'll leave as it's waiting for your reply.

Thank you

Frix-x, 2023-06-21:
Hey,

I've had a look at your klippy log. It's indeed pretty simple: I think you did not followed the install documentation until the end (the "Usage" part of the doc) as you don't call my macro but still the original BED_MESH_CALIBRATE macro in your PRINT_START (see following picture):

[image]

So the fix is pretty easy in your case as **you just need to replace it by by these two lines**:
```
{% set FL_SIZE = params.SIZE|default("0_0_0_0")|string %}
ADAPTIVE_BED_MESH SIZE={FL_SIZE}
```

Let me know if it's working correctly :)

R4toLoko, 2023-06-22:
@Frix-x ,

If I only had expanded this section . No comments....

[image]

Unless I missed that part but it does not mention that `BED_MESH_CALIBRATE` has to be replace with `ADAPTIVE_BED_MESH SIZE={FL_SIZE}`
I do have a suggestion, it might be worth having those lines added into the doc for those like me looking for the standalone macro without replacing the whole `printer.cfg`

I am printing a cube and it only did a mesh for the cube area, beautiful.
So I can confirm, a LDO Trident with stock `printer.cfg` with the macro and those two lines does the trick.

Thank you so much for the help :)

closed 2023-06-22 by R4toLoko