Frix-x/klippain #250: Idea: Improve heatsoak macro

crazy-daddy, 2023-06-30:
I stumbled across some neat ideas on how to improve the soaking process in the Klipper discourse and Github. 

Basically the soaking gets more dynamic i.e. no "wait for 20 minutes" or "wait for X°, oh wait it's winter, so I gt adapt my settings, as I will never reach that". Instead it's an interruptible (YAY!) "background task" in the beginning of the start-print sequence that checks if a significant change (X Kelvin) in temperature of sensor Y has been achieved within the last Z seconds and if not then it declares the soak to be ended. 

Would this be something of interest; i.e. should I try and optimize my "hacked together" solution to a (somewhat) readable PR for klippain?

Surion79, 2023-07-02:
hmm, so it is a chamber soak instead of a bed soak? I am not sure about this variable cancel of soaking. E.g. you forgot 3dlac, open the chamber and soaking ends in that logic

crazy-daddy, 2023-07-02:
The most important point for me was the interruptible part. Often started a print only to realize like 2 minutes later, that I needed to change sth. or whatever. Always doing an emergency stop or waiting for the whole startup routine to be finished is ... meh.

I have now "graduated" to using https://github.com/garethky/klipper-voron2.4-config/blob/mainline/printer_data/config/heatsoak.readme.md - who had the same idea as me, but is infinitely better at utilizing it. 

You are of course correct in your critique, that would be an issue! I'll have to dwell on that one... Maybe I'll get it to do a "double check" with my second bed thermistor... i.e. soak until that one is stable and THEN go for chambersoak. 

But I realize we're getting a bit too "niche" there - so maybe not a good addition to klippain.

Frix-x, 2023-07-03:
Hey thanks for raising an issue about that.

To be honnest, it was like that in the past (before this repo was renaimed to Klippain and became more a public things). The problem with the interuptible heat soak is more technical in term of Klipper handling Jinja and macros. The bahavior was quite complicated and leaded to weird things in some cases. Moreover, now since we have splited the `START_PRINT` macro to make it customizable it would be very complicated to handle that properly. I still keep it on my mind if I find a way to do it correctly without adding side complications.

However, the "adaptive" heat soak part is quite interesting and I've already made some testing about using the PWM value to wait for stabilization before continuing (instead of a fixed time). I just need to find some time to implement that properly but my testings were working pretty well :)

Surion79, 2023-12-11:
> The most important point for me was the interruptible part. Often started a print only to realize like 2 minutes later, that I needed to change sth. or whatever. Always doing an emergency stop or [...]

If you forget something you end the print. So I don't see the issue with using emergency button. If you set up chamber soaking, you can cancel the print but you keep the temp, so if you start again, the soak is not starting from the beginning and should reach faster your goal temp.

closed 2024-04-14 by github-actions[bot]