Frix-x/klippain #258: FR:  Sensorless homing only one axis

Glenn-Kerselaers, 2023-07-17:
At the moment it's not possible, using the Sensorless_homing.cfg, to enable Sensorless homing on only the X or y axis.

The .cfg file has to be commented out and Sensorless homing parameters has to be configured in override.cfg.

Maybe split the Sensorless_homing.cfg for each axis?

Frix-x, 2023-07-17:
Oh mhmm ok, this is possible.
I did not think about a case where the user would have different homing type for X and Y ^^ Can you explain your use case?

Glenn-Kerselaers, 2023-07-17:
The LDO Voron Switchwire Kit uses Sensorless homing on the y axis. But a micro switch endstop on the X-axis.

Surion79, 2023-07-17:
oof, that multiplies the config files considerably, taking the tmc2240 files into account. it would require for each axis and each tmc and implementation variants own files which has to be uncommented individually.

Frix-x, 2023-10-08:
This is now done alongside the new input shaper scripts that will be available in the following days (next Klippain release) :)

teookie, 2023-12-23:
Just popping in to say that I use sensorless X homing and limit switch Y homing on both my core xy printers. So, the ability to apply sensorless homing on a per-axis basis in Klippain would help my use case.

mjonuschat, 2026-08-15:
I'm closing this out, this should be achievable with a simple entry in overrides.cfg. Include the sensorless file for your driver as usual, then add an override for the axis that should stay on a physical endstop. Example for TMC2209 with sensorless X and a physical switch on Y:

printer.cfg
```name=printer.cfg
  [include config/software/sensorless_homing/sensorless_TMC2209.cfg]
  ...
  [include overrides.cfg]   # must stay last
```

overrides.cfg
```name=overrides.cfg
  [stepper_y]
  endstop_pin: Y_STOP
```

closed 2026-08-15 by mjonuschat