Frix-x/klippain #262: Switching off the bed soak when the chamber temperature is above 40 degrees.

AMFkuna, 2023-07-25:
Hi
If I want to print after finishing the previous print and the chamber is heated above 40 degrees, and the bed cools down to, for example, 50 degrees, then I have to wait for the bed soak time during the next print. It would be good to add a variable for example bedsoak_min_chamber_temp to enable bed soak up to a certain chamber temperature.
Thanks

Surion79, 2023-07-25:
Klippain has already a chamber soak timer. in your set up, i would set the bed soak timer variable to 0-1 and raise the chamber soak timer higher so it does not continue when time has passed but the chamber has the temperature.

Surion79, 2023-07-25:
(thinking about it, i should configure it as I wrote for myself :D )

Surion79, 2023-07-25:
[image]
Mine looks now like this. chamber soak is 25 minutes, chamber temp min 50 °C. no bed soak (set to 0) as it is controlled by chamber soak. I think that the default soak is required if the user has no chamber thermistor but wants to soak it by "thumb" via a mere timer.

Surion79, 2023-12-07:
functionality can be achieved by using the provided settings as example

closed 2023-12-07 by Surion79