Frix-x/klippain #267: Adaptive bed mesh returns error if there isn't a "exclude object"property

SinisterRj, 2023-07-31:
The new version (3.0) of the Adaptive bed mesh macro is returning this error if the gcode is not generated by a slicer that supports "exclude objects", like PA test ou any other test code generated by a browser:

`Error evaluating 'gcode_macro COMPUTE_MESH_PARAMETERS:gcode': jinja2.exceptions.UndefinedError: 'extras.gcode_macro.GetStatusWrapper object' has no attribute 'exclude_object'`

Frix-x, 2023-07-31:
Oh, I will need to work on that... But the thing is that I did not touched anything on that side, so this error could potentially be already there before v3 of the adaptive mesh.

I'll see how I can fix it :)

Surion79, 2023-07-31:
I added a PR to address this issue. That similar issue came up on the ercf restart check.

Frix-x, 2023-08-01:
And it's merged in the develop branch. Public release is planned soon :)

closed 2023-08-18 by Frix-x