Frix-x/klippain #272: Orbiter 2.0 configuration is wrong (in a potentially breaking things way)

blutack, 2023-08-04:
The Orbiter 2.0 configuration sets gear_ratio, but also uses the rotation_distance from the official documentation (which does *not* set gear_ratio).

This means that the rotation distance ends up being 7.5* what it should be, so the extruder tries to push 7.5 times as much flow as it should. As the orbiter is very good at gripping plastic and can exert a pretty serious amount of force the massively overloaded hotend will try and separate from the mount (in my case the hotend dismantled itself and made a bit of a mess).

I patched this locally by overriding to `gear_ratio: 1:1` and everything works as expected.

Thanks for klippain!

Frix-x, 2023-08-07:
Oh this is good to know. Thank you, I'll correct it as soon as possible :)

blutack, 2023-08-07:
Thanks! Let me know if you want me to test anything out.

Frix-x, 2023-08-07:
I removed the gear_ratio in this commit: 84aba96cce4aba8f3c53fd676fd36787b3944674
It's now on the develop branch and will be released in the next Klippain release soon :)

closed 2023-08-18 by Frix-x