Frix-x/klippain #278: Pin naming conventions for neopixels when multiple ports are present

ndanyluk, 2023-08-09:
Hi! So it's been a while since I was last working on #208, but one item that I never got around to resolving was the presence of duplicate features with independent pins on a mainboard.  

Specifically, the LDO picobilical includes 2 separate, independently controllable neopixel ports.  The intent, so far as I know, is to control a left-hand neopixel strip and a right-hand one.

Currently Klippain only supports STATUS_NEOPIXEL, LIGHT_NEOPIXEL, and LIGHT_OUTPUT.  It would be nice to change this to be something like LIGHT_NEOPIXEL[0,1,2,etc] across Klippain so users have a more standardized way of accessing these features.

Thoughts? The biggest issue I see is that if we make LIGHT_NEOPIXEL0/1 replace LIGHT_NEOPIXEL/unused, then it would be a breaking change for users of LIGHT_NEOPIXEL.

Frix-x, 2023-08-11:
> Currently Klippain only supports STATUS_NEOPIXEL, LIGHT_NEOPIXEL, and LIGHT_OUTPUT. It would be nice to change this to be something like LIGHT_NEOPIXEL[0,1,2,etc] across Klippain so users have a more standardized way of accessing these features.

I understand this but still not sure about how to deal it easily...
Here is my thoughts: Klippain control only one set of lights connected on a single neopixel pin at the moment. I don't think there is a real usecase to get like 2 different control pin since when you chain the strips you can still control all the leds independently in the same way.
So beside an "ease of wiring" it serve no purpose to use the two ports but only complicate things... Would it be possible to use only one port? So this means, defining both ports in the manufacturer boards pins, and then in the user mcu template using one or the other (and this way the user can use the second one for some custom things if he wants to in his own macros or whatever).

Regarding the filament runout sensor pins, I think Klipper support only one definition so using both pins can't work at the moment. But it can be done in the same way: defining both pins in the manufacturer file and then using one or the other in the user templates. The remaining pin can be used for something else by the user (or later by Klippain for a new feature)

ndanyluk, 2023-08-12:
Hi Frix, 

I understand and I actually hadn't considered only implementing it in the board config and not the user template, that's a great idea and would resolve the problem cleanest I think.

Unfortunately in this case, it's feasible but very undesirable to only use one neopixel port on the picobolical.  The design of the board and placement of the connectors is designed around these style LED strips with JSTs: https://west3d.com/products/rainbow-on-a-matchstick

It can be done, but I'd likely opt for the mcu file changes and just implement it in overrides in that case rather than change the led wiring.

Frix-x, 2023-08-15:
Yes the overrides are always possible to overcome this kind of problems at the end and is indeed a first step :)

closed 2023-09-03 by Frix-x