Frix-x/klippain #280: Missing [manual_mh_stepper selector_stepper] section in mmu_hardware.cfg (New HappyHare)

Fragmon, 2023-08-09:
I have an error with the combination orbiter2 and mmu with the klippain config.
I use the this branch https://github.com/Frix-x/klippain/tree/new-ercf_HHv2

https://user-images.githubusercontent.com/50713890/259345643-7b5f9b62-47a4-4e66-92f1-78356a8e99a0.png
[image]
[image]

I cant upload the printer cfg because i reset to stock klippain because i need the printer
@Benoitone

Benoitone, 2023-08-09:
Can you post your klipper.log?

closed 2023-08-16 by Frix-x

Fragmon, 2023-08-16:
Problem is solved with update happy hare branch :-) thx @Benoitone