Frix-x/klippain #282: Add BigTreeTech SFS V2.0 support

M3HNGRY, 2023-08-14:
This is just to track and not forget for the next version to add this: https://github.com/bigtreetech/smart-filament-detection-module

```
[filament_switch_sensor switch_sensor]
switch_pin: gpio_xx
pause_on_runout: False
runout_gcode:
  PAUSE # [pause_resume] is required in printer.cfg
  M117 Filament switch runout
insert_gcode:
  M117 Filament switch inserted

[filament_motion_sensor encoder_sensor]
switch_pin: gpio_xx
detection_length: 2.88
extruder: extruder
pause_on_runout: False
runout_gcode:
  PAUSE # [pause_resume] is required in printer.cfg
  M117 Filament encoder runout
insert_gcode:
  M117 Filament encoder inserted
```

Frix-x, 2023-08-15:
Hello!

Thanks for raising this issue! I will need to think about how to get it integrated properly since it's two pins that are needed for this sensor to work properly and currently Klippain only support the standard "one pin runout sensors" that are either a motion sensor or a switch sensor and not both at the same time.
However. I'm not sure about the real advatage of having both at the same time and suspect that only the motion part is enough to have something working correctly.

austinrdennis, 2023-08-21:
> Hello!
>
> Thanks for raising this issue! I will need to think about how to get it integrated properly since it's two pins that are needed for this sensor to work properly and currently Klippain only [...]

Hi! 

Maybe I can help clarify here on what makes this sensor different from the classic "one pin" runout sensors. This sensor uses an optical sensor for runout detection (filament spool is empty) and an encoder wheel for clog detection (filament is present but not moving properly through the nozzle) so you get both types sensors in one compact package. For the BTT Octopus board for example, it indicates how this device should be hooked up in it's [wiring diagram](https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-V1.0/blob/master/BIQU%20Octopus%20wiring%20diagram-V1.0-2021-7-9.png) (labeled "Smart Filament Sensor"). So those pins for the BTT Octopus would be PG_13 (J32)for the switch sensor and PG_12 (J28) for the motion sensor.

Both config sections are required at the same time to use the device as intended (and klipper supports this), and either sensor can trigger the M117 command independently of each other. If you really only wanted to use one for some reason, the motion sensor would be the one to pick. I have and use one of these sensors and it works great! I hope this helps, please fell free to ask me any questions!

austinrdennis, 2023-10-17:
> Hi @austinrdennis 
> 
> Do you use the stock Klippain pause_resume macros or do you use something like what Ellis recommends, https://ellis3dp.com/Print-Tuning-Guide/articles/useful_macros/pause_resume_filament.html#smart-filament-sensor, that includes disabling the sensors at startup and during the pauses? I would be interested to see the relevant portions of your configs.
> 
> Thanks

I'm using the standard Klippain Pause/Resume macros. I already worked with Frix to ensure the filament sensor is disabled during extruder movements that are done outside a normal print. That's not out quite yet, but it is coming next release. You should only need to disable the sensor for certain events and not the entire startup sequence.

I'll post a link to my config repo sometime today. It's not stock Klippain, but it's very similar. I just removed all the stuff that wasn't applicable to my printer.

LongDono, 2023-10-18:
I also got one of these two weeks ago, another example:

```[filament_switch_sensor switch_sensor] # 2023.09.28
#===================================================================================================
switch_pin: PG12
pause_on_runout: False
runout_gcode:
    RESPOND PREFIX=Sensor MSG="Switch detects filament runout"
    PAUSE # [pause_resume] is required in printer.cfg
    #M117 Filament switch runout
insert_gcode:
    RESPOND PREFIX=Filament MSG="Switch detects filament present"
    #M117 Filament switch inserted

[filament_motion_sensor encoder_sensor]
switch_pin: PG14
detection_length: 5 # Accuracy is 2.88mm at lowest, recommend increasing in 1mm increments
extruder: extruder
pause_on_runout: False
runout_gcode:
    RESPOND PREFIX=Sensor MSG="Encoder detects filament not moving"
    PAUSE # [pause_resume] is required in printer.cfg
    #M117 Filament encoder runout
insert_gcode:
    RESPOND PREFIX=Sensor MSG="Encoder detects filament movement"
    #M117 Filament encoder inserted

[gcode_macro _USER_VARIABLES]
variable_filament_sensor_enabled: True
gcode:
#---------------------------------------------------------------------------------------------------

ksteddom, 2023-10-21:
@austinrdennis @Frix-x Is there a particular branch that the fix is in? I would be happy to test it out.
Thanks

austinrdennis, 2023-10-21:
> @austinrdennis @Frix-x Is there a particular branch that the fix is in? I would be happy to test it out.
> 
> Thanks

I don't think anything has been pushed for this yet as far as including both sensor types.

Surion79, 2023-11-02:
I had an idea for integration, but first i need to understand it more. Which sensor takes precedence? As klipper knows two different types of sensors, you have two gcode "runout_gcode" sections independently.

austinrdennis, 2023-11-10:
> I had an idea for integration (make both sensor modules unique, so both can be activated for SFS), but first i need to understand it more. Which sensor takes precedence? As klipper knows two different types of sensors, you have two gcode "runout_gcode" sections independently. (and you need to override both sensors with the runout_gcode section)

Neither take precedence, they just run whatever is in the "runout_gcode" section when the sensor detects a runout event. Either one can trigger a runout event individually without regard to the other and they can be active at the same time. I have mine run a common macro from the runout_gcode section of each sensor, but they could do different things depending on which one is triggered.

The main barrier to using both right now isn't the config for the sensors (just include both types) but rather how the sensor is referenced in certain macros to deactivate the encoder during filament operations that will likely cause under-extrusion (ex. loading filament) and thus avoiding eroniuos runout event triggers. 

The path forward would be to name each sensor a unique name in the config section and only deactivate the encoder sensor in the macros section. Easiest way to do this in practice is to leave the encoder sensor named what it is currently and rename the switch sensor to something unique as it never needs to be deactivated. Then both configs could be active at once and no changes are needed for the macros.

agios, 2024-02-05:
This works almost out of the box, already using it on my setup. The only issue with the klippain config is that both pins are aliased as `RUNOUT_SENSOR` in the configs, so you can only enable one, and then you just need to add the other one in the overrides with the second pin config.

I guess that could be fixed by renaming one of the two to eg `MOTION_SENSOR`, adding a default alias from MOTION_SENSOR to RUNOUT_SENSOR to maintain backwards compatibility. If @Frix-x thinks this is useful I could try opening a PR.

Surion79, 2024-02-05:
The question would be what is a real usage scenario for having two sensor checks for 'filament issues' resulting in the requirement to have two different macros run.

agios, 2024-02-06:
I guess redundancy and more peace of mind? The important is the motion one, as it will detect both underextrusion and runout. Truth be told though if it fails it will probably fail as "no motion detected", so it should still be safe. But well, when you get hardware with two sensors, it feels weird not to connect and use both, OCD is a thing 😁

agios, 2024-02-08:
I was thinking about it @Surion79 and there is actually a good use case. In cases of materials like TPU, where you are now able to disable the motion runout sensor per material to avoid false underextrusion triggers, you would still want to keep the on/off runout sensor enabled to detect actual filament runout.

Surion79, 2024-02-08:
That is indeed a good use case.
That would require a bit more to disable encoder when material 'tpu' is used?

agios, 2024-02-08:
Now both switch and motion are configured to be called `runout_sensor` and scripts depend on finding the active one just by its name. As it is now, you can just enable the encoder using the builtin motion support, so that one becomes the "runout_sensor" which can be disabled per filament, and then you just add the switch sensor through overrides with a different name, should all just work in the expected way.

roykrikke, 2024-04-03:
Is there any progress on this topic? I want to use my SDS V2.0 with both sensors.

wight554, 2024-05-06:
I see it's still not compatible, so I made quick hack (works in case there's only 1 filament sensor on printer):
```python
[gcode_macro _USER_VARIABLES]
variable_filament_sensor_enabled: True
gcode:

[filament_switch_sensor switch_sensor]
switch_pin: ^RUNOUT_SWITCH_SENSOR
pause_on_runout: True
# runout_gcode:
#     PAUSE # [pause_resume] is required in printer.cfg
#     M117 Filament switch runout
# insert_gcode:
#     M117 Filament switch inserted

[filament_motion_sensor motion_sensor]
switch_pin: ^RUNOUT_MOTION_SENSOR
detection_length: 7
extruder: extruder
pause_on_runout: True
event_delay: 3.0
pause_delay: 0.5
# runout_gcode:
#     PAUSE # [pause_resume] is required in printer.cfg
#     M117 Filament encoder runout
# insert_gcode:
#     M117 Filament encoder inserted

[gcode_macro SET_FILAMENT_SENSOR]
rename_existing: BASE_SET_FILAMENT_SENSOR
description: Sets SFS v2 filament sensors
gcode:
    SET_FILAMENT_SENSOR SENSOR=motion_sensor ENABLE={params.ENABLE}
    SET_FILAMENT_SENSOR SENSOR=switch_sensor ENABLE={params.ENABLE}
 ```

roykrikke, 2024-10-25:
Kindly meant: Is there any progress on this topic beyond the quick hack?