Frix-x/klippain #283: Mismatch in SKR Pico mcu_definition and user_template files

NoSoucy10, 2023-08-18:
### Klippain branch

- [X] I confirm using the main branch

### Version

v4.1.0-0-g5e24c04f and earlier

### Describe the bug and expected behavior

There's a mismatch in the extruder motor pins' names between the SKR Pico mcu_definition file and the user_template that's used to generate the mcu.cfg file.

In the definition file, the extruder motor pins are named MCU_E0_* but in the template it uses MCU_E_*

user_template:
[image]

mcu_definition:
[image]

This causes an error with klipper that can't find the requested pins.
[image]

Renaming them to the correct ones in my mcu.cfg fixed the error.

I suggest renaming the ones in the SKR Pico user_template file to the ones in the mcu_definition file.
Sorry for not just making a PR for this simple issue, I don't really know how to participate in a github repo.

Midhy

Frix-x, 2023-08-21:
Thanks for catching this!
I'll fix it in the next release for everyone :)

Surion79, 2023-08-31:
Oooh, nice fiend from me too!
I wonder why this never came up on my pico. Need to check after move

Frix-x, 2023-09-18:
This should be fixed in 00ec18e8e89e5edb01f8351130d23c03b2f149c5 to be released in the next version :)

closed 2023-09-18 by Frix-x