Frix-x/klippain #284: Klipper reports system_info.py is missing but it really isn't

ksteddom, 2023-08-20:
### Klippain branch

- [X] I confirm using the main branch

### Version

v4.1.0-0-g5e24c04f  and previous release as well

### Describe the bug and expected behavior

I get the following message in the console when I start up Fluidd.
` 08:06:04  // Running Command {system_info}...:
08:06:04  // python3: can't open file '/home/pi/printer_data/config/scripts/system_info.py': [Errno 2] No such file or directory
08:06:04  // Command {system_info} finished
08:06:04  // Currently configured MCU(s):
// mcu: v0.11.0-240-g6d48adf9
// mcu toolhead: v0.11.0-257-ged66982b
08:06:04  echo: Klippain started and ready !
`
I checked that the file exists and is set as executable by user, group, and others (-rwxr-xr-x ).
Everything seems to be running fine so I am not sure where the error is coming from. This is a new build and has always been there so may very well be an issue with my configuration. 

### Additional information and klippy.log

[klippy.log](https://github.com/Frix-x/klippain/files/12388143/klippy.log)

Frix-x, 2023-08-21:
Hi,

Looking at your Klippy.log it's because you are not using the "pi" user and the correct path is `home/ksteddom/printer_data/config/scripts/system_info.py`. Unfortunately the code shell command extension doesn't allow me to use relative path here or alias like `~/` so this is needed to be customized manually.

So just override the two `[gcode_shell_command xxxx]` of Klippain by adding this to your overrides.cfg:
```
[gcode_shell_command plot_graph]
command: bash /home/ksteddom/printer_data/config/scripts/plot_graphs.sh

[gcode_shell_command system_info]
command: python3 /home/ksteddom/printer_data/config/scripts/system_info.py
```

ksteddom, 2023-08-21:
Thanks! Loving the software. Very well done.