Frix-x/klippain #285: Default park position is less than minimum in the size config causing "Movement out of range" error

ksteddom, 2023-08-20:
### Klippain branch

- [X] I confirm using the main branch

### Version

v4.1.0-0-g5e24c04f  and previous release as well

### Describe the bug and expected behavior

I got a movement out of range error the first time I ran my park macro. I looked at the size config file, https://github.com/Frix-x/klippain/blob/main/config/hardware/axis/size/350mm.cfg, and the minimum x and y values are set to 0. In https://github.com/Frix-x/klippain/blob/main/user_templates/variables.cfg the park position is set to
`variable_park_position_xy: -1, -1`.
It is an easy fix to either override the minimum x and y values in my overrides.cfg file or change the park position in the variables.cfg file but seems like the repo should have the default changed so there is not a mismatch.

### Additional information and klippy.log

[klippy.log](https://github.com/Frix-x/klippain/files/12388192/klippy.log)

Fragmon, 2023-08-21:
Thats not a bug. I think the -1, -1 is set to avoid damage. Because of this number you get an error and a" remember" to adjust the values.

Frix-x, 2023-08-21:
Hi thanks for raising an issue :)
This is indeed not a but ans done on purpose. The variables.cfg file is meant to be customized to your own machine. As every machine is different and because everyone need to park at a different position I prefer to set it to -1, -1 by default in order to produce an error and avoid crashing into something.

But this make me think about an improved mechanism to add for the next version: check on Klippain startup if there is still some unset variables like this park position and warm the user about it directly. This way it will fail before starting a print and can be correct directly.

ksteddom, 2023-08-21:
Thank you! I'll update those.

Surion79, 2023-08-31:
@Frix-x there is already a pr to check startup variables. I think this is a good place to implement it there

BenBergman, 2025-08-12:
For me, I'd like the PARK macro to only move up in Z and stay where it is in XY. This avoids crashing into other objects if I'm printing multiple parts one at a time and a there is already a completed part on the bed. Instead of a check to see if park x and y are set, would it be reasonable to assume park xy set to -1 means don't move in x or y?

BenBergman, 2025-08-12:
I've changed my `PARK` macro to set Px and Py as follows

```
...
    {% set Px, Py = printer["gcode_macro _USER_VARIABLES"].park_position_xy|map('float') %}
    {% if Px < 0 %}
        {% set Px = printer.toolhead.position.x|float %}
    {% endif %}
    {% if Py < 0 %}
        {% set Py = printer.toolhead.position.y|float %}
    {% endif %}
    {% set Px = params.X|default(Px)|float %}
    {% set Py = params.Y|default(Py)|float %}
...
```