Frix-x/klippain #286: Check probe is not working

Frix-x, 2023-08-28:
### Klippain branch

- [X] I confirm using the main branch

### Version

v4.1.0

### Describe the bug and expected behavior

Apparently the check probe is not checking anything here: https://github.com/Frix-x/klippain/blob/5e24c04f646d6be4d56ad71e96b47fdfb749a910/macros/base/probing/dockable_probe.cfg#L297-L301

I need to validate what is happening exaclty since this macro was written like that 2 years ago (since the 23th November 2021) and everything is working "OK" despite the error (not using `params.ACTION`)... 🤣

### Additional information and klippy.log

This was issued on Discord
[image]

Surion79, 2023-08-31:
So does it miss just the params.ACTION or is it more than just this?

Frix-x, 2023-09-03:
Yes it's a quick fix but I want to test and validate that the behavior is not changed (it should not) before merging

closed 2023-09-10 by Frix-x