Frix-x/klippain #287: No trigger on probe after full movement

GuillaumeP01, 2023-09-01:
### Klippain branch

- [X] I confirm using the main branch

### Version

4.0

### Describe the bug and expected behavior

Good morning,
impossible to operate the adaptive bed mesh macro.
With each impression, my head crashes into the tray.
I did all the bltouch configuration from this page: https://www.klipper3d.org/fr/BLTouch.html?h=bltouch
and everything is perfectly functional.
The G28 command works perfectly as well as the BED_MESH_CALIBRATE command.
It seems to me that I have correctly configured my printer.cfg 
[printer.txt](https://github.com/Frix-x/klippain/files/12501191/printer.txt)
.
Every time I run a print I get this:

`07:22:52
No trigger on probe after full movement
07:22:52
No trigger on probe after full movement
07:22:52
No trigger on probe after full movement
07:22:51
No trigger on probe after full movement
07:22:45
bed_mesh: zero_reference_position is (167.50, 167.50)
07:22:45
34 | (230.5, 223.0) | (228.0, 257.0)
07:22:45
33 | (201.2, 223.0) | (198.8, 257.0)
07:22:45
32 | (172.0, 223.0) | (169.5, 257.0)
07:22:45
31 | (142.8, 223.0) | (140.2, 257.0)
07:22:45
30 | (113.5, 223.0) | (111.0, 257.0)
07:22:45
29 | (113.5, 192.1) | (111.0, 226.1)
07:22:45
28 | (142.8, 192.1) | (140.2, 226.1)
07:22:45
27 | (172.0, 192.1) | (169.5, 226.1)
07:22:45
26 | (201.2, 192.1) | (198.8, 226.1)
07:22:45
25 | (230.5, 192.1) | (228.0, 226.1)
07:22:45
24 | (230.5, 161.3) | (228.0, 195.3)
07:22:45
23 | (201.2, 161.3) | (198.8, 195.3)
07:22:45
22 | (172.0, 161.3) | (169.5, 195.3)
07:22:45
21 | (142.8, 161.3) | (140.2, 195.3)
07:22:45
20 | (113.5, 161.3) | (111.0, 195.3)
07:22:45
19 | (113.5, 130.5) | (111.0, 164.5)
07:22:45
18 | (142.8, 130.5) | (140.2, 164.5)
07:22:45
17 | (172.0, 130.5) | (169.5, 164.5)
07:22:45
16 | (201.2, 130.5) | (198.8, 164.5)
07:22:45
15 | (230.5, 130.5) | (228.0, 164.5)
07:22:45
14 | (230.5, 99.7) | (228.0, 133.7)
07:22:45
13 | (201.2, 99.7) | (198.8, 133.7)
07:22:45
12 | (172.0, 99.7) | (169.5, 133.7)
07:22:45
11 | (142.8, 99.7) | (140.2, 133.7)
07:22:45
10 | (113.5, 99.7) | (111.0, 133.7)
07:22:45
9 | (113.5, 68.8) | (111.0, 102.8)
07:22:45
8 | (142.8, 68.8) | (140.2, 102.8)
07:22:45
7 | (172.0, 68.8) | (169.5, 102.8)
07:22:45
6 | (201.2, 68.8) | (198.8, 102.8)
07:22:45
5 | (230.5, 68.8) | (228.0, 102.8)
07:22:45
4 | (230.5, 38.0) | (228.0, 72.0)
07:22:45
3 | (201.2, 38.0) | (198.8, 72.0)
07:22:45
2 | (172.0, 38.0) | (169.5, 72.0)
07:22:45
1 | (142.8, 38.0) | (140.2, 72.0)
07:22:45
0 | (113.5, 38.0) | (111.0, 72.0)
07:22:45
bed_mesh: generated points
Index | Tool Adjusted | Probe
07:22:45
Generating new points...
07:22:45
echo: Adaptive bed mesh: MESH_MIN=111,72 MESH_MAX=228,257 PROBE_COUNT=5,7 ALGORITHM=bicubic
07:22:39
echo: Computed mesh parameters: MESH_MIN=111,72 MESH_MAX=228,257 PROBE_COUNT=5,7 ALGORITHM=bicubic
07:22:39
echo: Got a SIZE parameter for the adaptive bed mesh
07:22:39
echo: Adaptive bed mesh: parameters not already computed, automatically calling the COMPUTE_MESH_PARAMETERS macro prior to the mesh
07:12:43
File selected
07:12:43
File opened:Calibration du debit_0.2mm_PETG_X1_26m.gcode Size:672707`

Where does the problem come from?
Thank you for your help
Guillaume

Frix-x, 2023-09-03:
Hi,
Thanks for raising an issue. I have to admit that I currently have no clues... Can you give me a full klippy log just after the error ?

GuillaumeP01, 2023-09-04:
Hi,
And here

[crowsnest.log](https://github.com/Frix-x/klippain/files/12516550/crowsnest.log)
[klippy.log](https://github.com/Frix-x/klippain/files/12516551/klippy.log)
[moonraker.log](https://github.com/Frix-x/klippain/files/12516552/moonraker.log)

Surion79, 2023-11-04:
if you add in your slicer in the starting GCode the G28 and BED_MESH_CALIBRATE before calling START_PRINT it works too? trying to get my head around it

Surion79, 2023-11-04:
In your G32 I highly suggest adding a G28 Z after QGL. I would also test removing loading the bed mesh in your print_start macro

Surion79, 2023-11-04:
and `position_min = -0.1` in stepper_z  is very tight.

Surion79, 2023-11-04:
all the above combined could lead to your issue.

Klippain uses a default position_min of -5

closed 2024-02-28 by github-actions[bot]