Frix-x/klippain #288: Add CUSTOM_MODULE to END_PRINT

AMFkuna, 2023-09-03:
### Describe the feature or hardware support you'd like

Hi 
Can you please add CUSTOM_MODULE to END_PRINT macro like in START_PRINT macro? I am using a clean brush which is located on the gantry and I would like to use it at the end of the print. Thanks

Frix-x, 2023-09-03:
Hello, thanks for the suggestion!

However i'm unsure about the usefullness of this: indeed the custom modules in the start print routine are here to be able to change the order of actions and/or add custom actions to a pretty complex macro.
In the case of the print end, the macro is pretty light and you should just override it and add your needed gcode lines directly

AMFkuna, 2023-09-04:
Hi,
You are right, it is simply to override this macro.

I found duplicity in END_PRINT macro variables definition
[image]

Benoitone, 2023-09-05:
Yes, for the duplicate... it's only in the Happy Hare branch that's in work in progress, in parallel of Happy Hare v2

closed 2023-09-10 by Frix-x