Frix-x/klippain #295: Filter stop during next print

AMFkuna, 2023-09-13:
### Klippain branch

- [X] I confirm using the main branch

### Version

v4.1.1

### Describe the bug and expected behavior

Hi
After printing is finished, the filter runs according to the set time. If I start repeated printing at this time, the filter runs according to the START_PRINT macro, but it turns off after the previous run-down timeout.

Frix-x, 2023-09-14:
Hey thanks for raising this issue. This is something I already detected. It should be a quick fix. I'll add it to the next version :)

Frix-x, 2023-09-18:
This is now fixed in the develop branch (commit 82e3ba8209944b1194a77355220d63c459ca86f3) and will be released in the next version :)

closed 2023-09-18 by Frix-x