Frix-x/klippain #298: Possible to drag nozzle across bed when homing X or Y axis individually.

austinrdennis, 2023-09-15:
### Klippain branch

- [X] I confirm using the main branch

### Version

V4.1.1 release

### Describe the bug and expected behavior

As I understand it, It's possible to drag nozzle across bed when homing X or Y axis individually when the Z axis is homed and the Z axis height is lower than the "homing_zhop" value due to a lack of checks in the XY homing logic.

I'm submitting a PR to fix this and I've tested it on my machine. I'm not a Klipper guru and I could be making a mistake. Since we're messing with motion safety, I'd like confirmation testing from someone before you go to merge it.

Surion79, 2023-11-06:
The issue is mentioned in the PR. With this change, the z axis would be marked as homed and will have an impact with non successful prints due to Conditional G28.

austinrdennis, 2023-11-10:
> The issue is mentioned in the PR. With this change, the z axis would be marked as homed and will have an impact with non successful prints due to Conditional G28.

When you cancel a print, why would you only home two of the three axes? They would all already be homed and conditional G28 would not attempt to re-home any axis. I'm genuinely confused on how this situation could reasonably occur. I just can't see a scenario where only 2 of the axes need to be homed before a print but not Z.

We can't save the user from themselves, but the z-height checks I ran into during normal use and nearly ruined my bed. I was homing XY repeatedly to calibrate sensorless homing without homing Z as that would have had high risk of crashing the nozzle outside my bed area.

austinrdennis, 2023-11-10:
Ok I understand now, see my comments on the PRs regarding this.

austinrdennis, 2023-11-11:
@Frix-x I think we came to a natural stopping point on this one and it's up to you to make a judgment call on whether to implement PR #299 or leave it as is.

Surion79, 2023-12-02:
currently on Hold until the PR on klipper is "done"

dpeart, 2023-12-08:
This happens to me, then I get a CAN bus timeout during QGL.  When I restart it drags the head across the deck as it homes X,Y.