Frix-x/klippain #311: Support for BTT EBB2209 using the RP2040 chip

dpeart, 2023-09-29:
### Describe the feature or hardware support you'd like

I'm using the BTT board, but the pinouts are completely different.  They are defined here:

https://github.com/bigtreetech/EBB/blob/master/EBB%20SB2209%20CAN%20(RP2040)/Hardware/EBB%20SB2209%20CAN%20V1.0%EF%BC%88RP2040%EF%BC%89-Pin.png

If we could get a configuration template added for this, it would be great.
thanks

dpeart, 2023-09-29:
https://github.com/bigtreetech/EBB/blob/master/EBB%20SB2209%20CAN%20(RP2040)/sample-bigtreetech-ebb-sb-rp2040-canbus-v1.0.cfg

Benoitone, 2023-09-29:
I think I have already made it… I will check very soon…

Benoitone, 2023-09-29:
@dpeart can you check?
https://github.com/Frix-x/klippain/pull/312

dpeart, 2023-10-06:
Is there something I need to do to get this update?  I tried today and it wasn't included in the download.

thanks

Frix-x, 2023-10-07:
Yes, it's still not on the main branch and will be released alongside the next version of Klippain planned very soon (hopefully this weekend, or beginning of next week)

dpeart, 2023-10-25:
I pulled the develop branch and got the mcu toolhead working with the new config file.  I only had to change one thing: the MCU_PROBE was set to some fan input.  I changed it to MCU_PROBE1

Everything else worked as expected.
thanks

closed 2023-11-17 by Frix-x