Frix-x/klippain #326: 3 Ideas for Klippain

Kiiiddd, 2023-10-26:
### Describe the feature or hardware support you'd like

So I am liking my Klippain install now that I have everything setup properly(my overides page is way bigger than I thought it was gonna be lol)

But I had a couple Ideas, not sure how viable or hard they would be to implement
Ellis Bed Fans module - I seen the clossed issue but I think it would be easy to add a different CFG that you could choose instead of Nevermore with Ellis's bed fan macro
Brush Safe Z height - My nozzle brush is pretty low and I had a nozzle scrape after a manual macro brush, and it probably wouldn't be hard to add a safe Z movement to the end of the Nozzle brush macro?
Multi-Color Print start & end - For doing Z-hop multicolor prints like fancy Voron faces, people will Z-hop around a different color but to do this you need to keep the mesh and load the current mesh without Homing&QGL. then you just modify your start and end G-code to use the other macros(https://docs.vorondesign.com/community/howto/mikhail/multi-colour-prints-with-a-single-nozzle.html). So having a 2nd Print Start Macro and End Macro than never gets used unless the G-code calls for it would be awesome

Frix-x, 2023-10-27:
Thanks for putting your request here. It will be easier for us to track them and add them in a futur release :)

Surion79, 2023-11-02:
- IIRC regarding the Bed Fans, it is tricky to implement, because at some point, you can't put everything in one function. Meaning you can't override and combine all additional functionality in base functions (like M190) there. That's why i dropped the Bed Fan integration. 
- how did you do a manual brush clean? (don't have one, so i don't know)
- Mmmh, wouldn't it be better to just add an additional parameter for start/end_print? Would be better to maintain instead of a second set of start/end_print macros

Kiiiddd, 2023-11-02:
- I have no idea as I am a idiot when it comes to coding, I just know I copy pasted the bed fan macro into the overrides and it hasn't broke anything. I thought it might work if it was a pick only one between the bed fans and nevermore. But again I have no idea so I will just still consider myself a idiot and be good lol
- By manually calling the macro from the dashboard. I have changed brush holder since then but the old one worked best at -0.5mm Z. It worked fine when doing a print start but after manually hitting the macro on the dashboard it stayed at -0.5mm Z and then I did something without thinking as it always worked fine in the print start and ended up with a bed scrape. I thought adding a safe Z movement to the end of the brush macro would prevent accidents
- I truefully don't know what would be the easiest. I went with Klippain because all the macros were pre setup. I have no problem with hardware and figuring out values for stuff makes sense to me, but looking at making macro's made me doubt life a bit lol. Just figuring out the right start G-Code in Orca Slicer took like an hour no joke

Surion79, 2023-11-02:
No worries, we are here to help :)
What is the name of the macro you used for the brush cleaning? Since I don't have it, it is easier for me to see where it could be improved, so I can add a variable and the corresponding fix.

Kiiiddd, 2023-11-02:
It was the gcode_macro CLEAN_NOZZLE included in Klippain. I used the dashboard to call it without any print running and at the end it left the nozzle at the brush height which is fine in most cases but with that brush holder installed it was low(I was using the old decontaminator mod which is pretty popular) the next action I took scraped the nozzle

Surion79, 2023-11-02:
do you expect just a specific retraction after nozzle clean?
can you write here what your brush_xyz  in user variables is?

Kiiiddd, 2023-11-02:
It's different now so I couldn't tell you but the Z height was what mattered any that was at -0.5mm( o just below Bed height). I always seen the nozzle clean during the print start be fine and didn't think anything of it when I manually called for the macro.

I think a +5mm Z movement at the end of the brush clean macro would stop any accidental scraping. If you think this is a bad thing do to adding any sort of time to the print start macro then feel free to ignore me as I changed my brush holder and now my nozzle clean height is +3mm if I remember right so it doesn't bother me anymore. Just thought if it would save 1 other person scraping the bed when using Klippain it would be good

Surion79, 2023-11-03:
@Frix-x : I would either write a warning if z is <0 or add a "zhop" to save z height

W141-ID, 2023-11-03:
Regarding the bedfans here is my working solution. Not the prettiest but it does work:

```
[gcode_macro START_FILTER]

[fan_generic bed_fans]
pin: BED_FAN
max_power: 1.0
kick_start_time: 0.250
off_below: 0.30
# hardware_pwm: True
# cycle_time: 0.001

gcode:
    {% set SPEED = params.SPEED|default(1)|float %}

    {% set filter_name = printer["gcode_macro _USER_VARIABLES"].filter_name %}
    SET_FAN_SPEED FAN={filter_name} SPEED={SPEED}
    SET_FAN_SPEED FAN=bed_fans SPEED={SPEED}

[gcode_macro STOP_FILTER]
gcode:
    {% set filter_name = printer["gcode_macro _USER_VARIABLES"].filter_name %}
    SET_FAN_SPEED FAN={filter_name} SPEED=0
    SET_FAN_SPEED FAN=bed_fans SPEED=0

[delayed_gcode _STOP_FILTER_DELAYED]
gcode:
    STOP_FILTER
```

They behave the same as the filter which I don't think is a problem.

Surion79, 2023-11-04:
These are just start-stop macros. The original bed fans macros are completely automated and are linked to the core bed heating functionality of klipper. you don't start/stop them. they work autonomously.
If you use a filter solution, you do have a bed fan solution implicetly, which is similar to yours. e.g. i use 3 Nevermore filters with just one pin and run them simultaneously. no need for bedfans, since it has 6 fans running. If you the filter with the side filters, you have already 4 fans, which are also working for heating chamber. 
Do you have a chamber thermistor? i use chamber soak to get to my temperature.