Frix-x/klippain #331: Fresh install does not check for all modules (e.g. matplotlib)

Surion79, 2023-11-05:
### Klippain branch

- [X] I confirm using the main branch

### Version

current

### Describe the bug and expected behavior

I tried after a fresh install via install.sh the belt tension function and it errored out after the run with 

> Traceback (most recent call last):
> File "/home/pi/printer_data/config/scripts/is_workflow/is_workflow.py", line 43, in
> from graph_belts import belts_calibration
> File "/home/pi/klippain_config/scripts/is_workflow/graph_belts.py", line 20, in
> import optparse, matplotlib, sys, importlib, os
> ModuleNotFoundError: No module named 'matplotlib'

seems a module is missing when installing

Surion79, 2023-11-05:
i added the module by apt install python3-matplotlib. it installed dozens of addditional packages. not sure if this is the solution for everyone

Frix-x, 2023-11-06:
This is part of the process of Klipper installation when doing resonances measurement. It's indeed needed.
But now this is installed by default on MainsailOS so it's not hit by a lot of people. I'm not sure that it's really needed to be added during the install script...

Surion79, 2023-11-07:
There was very recently a lot of ruckus about crowsnest being not compatible with Bookworm. (Had the same issue, so i tracked it)
First mainsail crew declined making it compatible by refering to using the older OS but changed their stance after even more queries on defect crowsnest rose up.

Especially since KIAUH does that stuff (installing required apt packages) when installing functionalities, there is no 'need' for mainsailOS.

Crowsnest even compiles binaries for multiple references in their install script, so i suggest adding it or giving a warning in klippain.
I am not even sure that it is that complicated to add an info/question ('for the resonance feature you need to install package xyz, do you want to do that now?'), ask for sudo and execute an apt install command.

Frix-x, 2023-11-29:
Ok this will be solved soon as I'm working on installing the Shake&Tune module in its own python venv to manage its libs dependencies. This is needed since each OS have their own package version and it's not reliable....

In the meantime, I'll rework how Shake&Tune is integrated into Klippain and migrating this part outside the codebase to be able to use the proper Shake&Tune install script and follow its updates independently :) It will be much easier to maintain!

closed 2024-02-21 by Frix-x