Frix-x/klippain #334: python3: can't open file '/home/pi/printer_data/config/scripts/system_info.py': [Errno 2] No such file or directory

EraserFX, 2023-11-07:
### Klippain branch

- [X] I confirm using the main branch

### Version

v4.1.1-1-gb49bcbfb

### Describe the bug and expected behavior

Mainsail Console:
python3: can't open file '/home/pi/printer_data/config/scripts/system_info.py': [Errno 2] No such file or directory

my user folder is not pi. how can i change it?

### Additional information and klippy.log

[klippy (7).log](https://github.com/Frix-x/klippain/files/13287766/klippy.7.log)

mUbsta, 2023-11-07:
Add the following to ~/printer_data/config/overrides.cfg replacing <username> with the user folder name

```
[gcode_shell_command system_info]
command: python3 /home/<username>/printer_data/config/scripts/system_info.py
```

You might want to add this at the same time.

```
[gcode_shell_command plot_graph]
command: bash /home/<username>/printer_data/config/scripts/plot_graphs.sh
```

Surion79, 2023-11-07:
it is fixed in the next version of klippain. Then the setting is independent of the user name.

closed 2023-11-17 by Frix-x