Frix-x/klippain #336: Orbiter Filament Sensor macro support

DrFate09, 2023-11-09:
### Describe the feature or hardware support you'd like

Does the current filament sensor macro(s) and Klippain in general support the Orbiter Filament Sensor pre-built macro configuration provided by the creator?

On the creators page for the sensor, he has shared with the community his own OrbiterSensor.cfg that provides several macros and variables for the user to use with their Orbiter Filament Sensor.

What would the best way of implementing this in Klippain be? Does it just involve overrides or something custom to be implemented? I figured this is a common macro for individuals who are using the Orbiter Filament Sensor and might be a popularity question.

### Additional context or information

References:
**Orbiter site**: https://www.orbiterprojects.com/orbiter-filament-sensor/
**OrbiterSensor.cfg (v2.2.03):** https://www.orbiterprojects.com/download/filamentsensor/OrbiterSensor.cfg

Surion79, 2023-11-10:
The easiest integration is to add the cfg in the overrides.cfg, overwriting the existing Klippain pause/resume macros. 
Either way, you loose functionality:
- Klippain: LED integration of Orbitor Sensor is not working by default for pause/resume
- Orbitor: Case Light integration in Klippain is not working as it is overwritten. (doesn't matter if you have no case light set up)

For me, the copyright info is missing in the file and on the project site. Also it is not hosted on Github. So I am not sure how to add this to klippain.

Frix-x, 2023-11-10:
Hello @DrFate09,

I've had a look at these macros and config files. They are basically some LOAD and UNLOAD macros with some variables to activate/deactivate the runout feature or tweak the filament length, etc.. during the LOAD or UNLOAD.

So basically, almost all of this is already managed by Klippain macros and doesn't need to be added: just activate the `switch_sensor.cfg` and you will be good to go. This is what I'm doing on my Prusa MK3s that is equipped with a very similar sensor.

The only part missing currently is the button management and the "autoload" but it can be added very easily using a small override like this (this is what I've got on my Prusa):
```toml
[gcode_button filament_unload_button]
pin: !Pxxx
press_gcode: UNLOAD_FILAMENT # Call Klippain stock UNLOAD macro on a button press

[filament_switch_sensor runout_sensor]
insert_gcode: LOAD_FILAMENT
```

DrFate09, 2023-11-10:
@Frix-x Thank you for your response, super useful. Any suggestions on how I would go about fixing the two missing functions that @Surion79 mentioned?