Frix-x/klippain #339: Nozzle Change

sirosimo, 2023-11-12:
### Describe the feature or hardware support you'd like

Hi, 
would it be possible to add some macros or support for nozzle changes ? 
Right now it requires messing with a lot of parameters to get printer setup after each nozzle changes (nozzle size, advance pressure,...). It may even have an impact on more tuning.
It would be great to have preset for each nozzle size and that it somewhat correlate with the material being used to use the right parameters.
I know Klipper will require a reboot but being able to run a macro for the correct nozzle size then reboot would be awesome.

Thank you 👍 

### Additional context or information

I found this [repo](https://github.com/garethky/change-nozzle-klipper-extra) that may be useful ? 
I found it through this [post](https://klipper.discourse.group/t/change-nozzles-without-a-restart/5964/11)

Surion79, 2023-11-14:
wouldn't a nozzle diameter change require gcode recalculation?
If you use a 0.8 nozzle and then change to a 0.4 nozzle with the same gcode file, i except at least artifacts in the print.
gcode recalculation is afaik not possible by just writing some macros. You would need some reprocessing to recalculate speeds, or at least extrusions of the loaded gcode file of the object.

The material settings in klippain are just setting some basic variables like filter speed, (un)retraction values, pressure advance, smoothing and z_offset. And the retract values are only used if firmware retraction is activated. this means that the slicer does not calculate/include these values. So no setting in material would have some sort of "extruder ratio" for a nozzle diameter change.

But maybe i am getting your question wrong. Do you have more insight/background on the reason/necessity?
For me your descriptions sounds more like that you need to create an additional slicer printer/nozzle configuration to easy switch to get the newest values for PA etc. in your slicer of choice, so you have the correctly fine tuned gcode for the correct diameter.
IIRC nozzle diameter also affect line width, which is procentually used in calculation of other widths. For me it looks the best option is to use the correct place to set these things up. It looks like you create something with mismatching settings and try to correct them in a post processing rountine instead of creating it correctly in the first place.
Or i don't understand your use case.

sirosimo, 2023-11-15:
Changing nozzle size would definitely start at the slicer level. So the gcode will be created based on the nozzle size.
But on the printer side, there is some considerations about what nozzle size is being used. 
It also would be nice to be able to set the nozzle size and somehow for the printer to record this value. So the next time I run a gcode, it checks if it is the correct nozzle size (the nozzle size is in the gcode).
But more importantly, the Advance Pressure and some other tuning will have to be changed as well. And this is not in the gcode. 

One work around I have currently is to use the MATERIAL argument from PRINT_START. In my slicer, I concatenate the material type and the nozzle size, then create the different profiles in my variables.cfg. That way I can create tuning profiles based on the nozzle size and filament type (for advance pressure, retraction,...) without having to mess with the overall config at each nozzle change. 

Some macros to make sure the nozzle is of the right size would be nice though

Surion79, 2023-11-15:
Looking at klipper i see a very old PR about setting diameter through gcode, but it was closed. It needs to be changeable by gcode to be used in macros. Klippain does not provide 'klipper extras' and the author of the linked repo describes his 'extra' as brittle. I think you already created the solution. For the check, you can override a custom module in klippain to check for nozzle diameter and put it at the very beginning of the startup sequence. For such topics the custom modules have been created.

closed 2024-02-28 by github-actions[bot]