Frix-x/klippain #340: First layer inconsistencies with machines that use DELTA_CALIBRATE, Z_TILT_ADJUST, or QUAD_GANTRY_LEVEL

austinrdennis, 2023-11-12:
### Klippain branch

- [X] I confirm using the main branch

### Version

v4.1.1

### Describe the bug and expected behavior

**Expected behavior:** First layer distance is consistent with no mechanical change to the printer on machines that use quad gantry level.

**What's going wrong:** During a quad gantry level procedure, the actual height of the nozzle can, and usually does, change slightly. This new change in position in never accounted for in Klipper, so the z height the printer thinks it's at remains the same and you end up with an unexpected offset (either positive or negative). The worse the gantry was misaligned prior to leveling, the more off it becomes after leveling. It ranges from 0.01mm to 0.25 mm for me depending on how well the gantry was aligned prior to QGL. I suspect it similarly affects all machines that level the bed to the gantry automatically, but I only have a QGL machine to test with.

**Solution:** Add "G28 Z" directly after leveling the gantry in the "start_print" macro or in the " _TILT_CALIBRATE" macro. This completely solved my issues and has been working like a dream for around 20 prints consecutively and through 2 nozzle changes. 

I'd love to submit a PR for this, but I'm not able to make multiple forks at the same time and I'm waiting for another one of my PRs to be resolved first.

### Additional information and klippy.log

**Additional Info:** This is a bit of "tribal knowledge" (knowledge that experienced people know, but isn't written down) in the klipper community and you really have to go looking for it to find any reference to it. Here's some resources I used to come to this conclusion in the first place, but you have to connect the dots and read between the lines a bit. There are very few places that this rule is mentioned specifically.

- Klipper config reference on [Calibrating Probe Z-Offset](https://www.klipper3d.org/Probe_Calibrate.html?h=calibrating+z+offset#calibrating-probe-z-offset):

> If the probe has an X or Y offset and the bed tilt is changed (eg, by adjusting bed screws, running DELTA_CALIBRATE, running Z_TILT_ADJUST, running QUAD_GANTRY_LEVEL, or similar) then it will invalidate the results of PROBE_CALIBRATE. After making any of the above adjustments it will be necessary to run PROBE_CALIBRATE again.

I read this as the Z-offset can change after tilting procedures.

- Ellis' Print Tuning Guide on [troubleshooting Z-offset inconsistencies](https://ellis3dp.com/Print-Tuning-Guide/articles/troubleshooting/first_layer_squish_consistency_issues/z_offset_inconsistency.html):

> For V2:
:warning: Ensure that you are homing Z again after QGL, as QGL throws off Z height. 

- Voron Design Community Documentation on [Z-Endstop Configuration](https://docs.vorondesign.com/community/howto/120decibell/z_endstop_configuration.html):

> 4. Run a QUAD_GANTRY_LEVEL to level the gantry.

> 5. Run a G28 again to home and center the nozzle, and establish a new known Z value. The nozzle should be at the center of the bed or at whatever point is used for reference for the bed mesh.

Frix-x, 2023-11-14:
Hello,

The homing after tilting you're referring to is already integrated into Klippain but a little bit differently. Indeed, it was previously made as you suggested, with a `G28 Z` command following immediately after QGL/Z_TILT. But this sequence was later modified during the transformation of the `START_PRINT` macro into distinct modules, ensuring a more efficient process flow. However, I assure you, this feature has not been removed.

Indeed, in the current setup, the homing process you're referring to is executed directly within the Z_offset module (that is called after the tlting process in all the default sequences). This change was implemented to minimize redundant homing operations during `START_PRINT`, especially for Voron TAP or this kind of probes that need a lower temperature to be able to probe without damaging the bed. This change was made in order to avoid the need to modify the temperature too many times: it's now done only once during the final Z offset adjustement.

Given this, it seems your issue might be stemming from a different source. To help diagnose the problem more accurately, could you please share your `variable_startprint_actions` configuration if you've made any customizations? Alternatively, if you're using the default settings, information about your probe type would be helpful to retrieve the sequence that is running in your machine.

austinrdennis, 2023-11-14:
> Hello,
>
>
>
> The homing after tilting you're referring to is already integrated into Klippain but a little bit differently. Indeed, it was previously made as you suggested, with a `G28 Z` command [...]

Wow, I can't believe I missed that module.

I've been running an override start-print sequence for a while and I didn't have the Z-offset module in there. I must have totally missed the boat on that.

Sorry to waste your time, I'll close this.