Frix-x/klippain #362: Add support to FYSETC Catalyst to Voron V0.2

pireybr, 2023-11-24:
### Describe the feature or hardware support you'd like

Please add support to the new MCU from Fysetc https://github.com/FYSETC/Catalyst_Kit

Frix-x, 2023-11-24:
Thanks for opening an issue, this will allow us to track it and include it in the next Klippain update

pireybr, 2023-12-03:
> @pireybr I took a look at it. Can you verify my assumption, that this is a mainboard with a breakout board as toolhead board? I see only a mainboard with pins. the toolhead pins are just breakouts from the major connector on the mainboard.

Exactly Surion, and there an accelerometer in the toolhead board, but it uses the same pins in the main board

closed 2023-12-06 by Frix-x