Frix-x/klippain #363: Manta M8P V2.0 SPI Pins Incorrect

Killajoedotcom, 2023-11-24:
### Klippain branch

- [X] I confirm using the main branch

### Version

v0.12.0-10-gea2f6bc0

### Describe the bug and expected behavior

When attempting SPI w/ 2240 drivers, there keep getting an error.
After investigating and trying multiple things, I found the SPI settings do not match the docs from BTT. I have edited the manta.cfg file and attached it for you to check.

[BTT_Manta_M8P_v2.0 - Copy.txt](https://github.com/Frix-x/klippain/files/13461825/BTT_Manta_M8P_v2.0.-.Copy.txt)

Surion79, 2023-11-29:
Can you please link the file you used for referencing? I mean the pinout

Killajoedotcom, 2023-11-29:
> Can you please link the file you used for referencing? I mean the pinout

Sure thing. It's on the BTT github located here:
https://github.com/bigtreetech/Manta-M8P/blob/master/V2.0/Firmware/generic-bigtreetech-manta-m8p-V2_0.cfg

Surion79, 2023-11-29:
have you connected the extruder connector to Motor5? The linked cfg points extruder to motor 5 and the pins are correct. there are two mcu files, the ones matching the real pins to aliases and the part where they are matched. The second file innterconnects extruder with motor7, but these user mcu.cfg is edited in every case, since the configuration there is just an example.
Or are you referring to something different?

Killajoedotcom, 2023-12-01:
@Surion79 The extruder is running via CAN bus. I understand how Klippain files interconnect and reference one another. The issue is different.

The linked BTT cfg is the stock base Manta M8P V2 config from BTT. My issue lies with the SPI pins. In Klippain they are listed as PC10, PC11, PC12. That configuration did not work with tmc2240s on motors 4, 5, and 6. Reference the tmc2130 section of the BTT cfg 

When I updated the overrides.cfg to use SPI pins PG6, PG7, & PG8 the SPI communication finally worked and now I have no errors referencing the DRV config. All motors move as they should.

Surion79, 2023-12-02:
Hey, thanks for the update. i found the issue. The mcu template is missing "your" SPI pins, the motor spi pins. The SPI pins PC10, PC11, PC12 are also SPI pins on the mainboard, but not defined for motor values apparently.

@Frix-x The referenced template is incomplete and has a missing, maybe wrong naming. how to proceed?

Frix-x, 2023-12-08:
So is it an error in Klippain? Or is it like two distinct bus?

Killajoedotcom, 2023-12-08:
> So is it an error in Klippain? Or is it like two distinct bus?
> 
> But if something is missing, we can add it directly to the manufacturer file and it will be available to everyone directly

Yes, it is both an error in Klippain and two distinct busses.

The SPI bus for motors are not the same as the SPI pins allocated in the Klippain config

Surion79, 2023-12-08:
yeah, but the wording is mixed IIRC, meaning the label is already wrongly used. I can put up a draft PR for verification

Frix-x, 2023-12-13:
I've just merged it to the develop branch. It will be available in the next Klippain release :)

Killajoedotcom, 2023-12-13:
The pin names in axis/ TMC/TMC5160 or TMC2240 need to be updated to the new changes as well. 
Right now they still refer to the wrong SPI bus instead of the new motor bus added. You can see from the copy/paste below that the spi pins need the "M" inserted.

[tmc5160 stepper_x]
spi_software_sclk_pin: DRIVER_SPI_SCK
spi_software_mosi_pin: DRIVER_SPI_MOSI
spi_software_miso_pin: DRIVER_SPI_MISO

Frix-x, 2023-12-13:
No this is right since it's referencing the user template that was updated as well:

1. Manufacturer file define this:
[image]

2. Then the user template file (that you may need to update on your side since it's an user file) is defining this:
[image]

So the pin aliases in the TMC5160 (ot the other TMCs) are corrects

Killajoedotcom, 2023-12-14:
My mistake. Thanks for pushing the updates.

closed 2023-12-14 by Killajoedotcom