Frix-x/klippain #364: Support for Filamentrix

GlnSevi, 2023-11-25:
### Describe the feature or hardware support you'd like

Hey,

I have problems to perform a filament cut. It always performs Tip Forming instead. I have another machine were Filamentrix runs smoothly. Can u implement it in your macros
Have a great day 😊

Surion79, 2023-11-29:
For that we need more information.
Probably sonething overwrites an exisiting custom macro.
We can't work against custom macros. Can you add your klippy log?

Frix-x, 2023-11-29:
I think this is probably not needed as it's already supported directly in HappyHare.
But I could miss something...

@Benoitone, do you have an idea about this?

Surion79, 2023-11-29:
i think his issue is that the machine is doing something different when using the function. so i think it is more a macro collision than a "feature request"

closed 2024-02-21 by Frix-x