Frix-x/klippain #382: Sherpa mini extruder with moons motor

EricZimmerman, 2023-12-07:
### Describe the feature or hardware support you'd like

Sherpa mini extruder with moons motor

my working config below (rapido v1 hot end)

### Additional context or information

[extruder]
step_pin: can0:E_STEP
dir_pin: !can0:E_DIR
enable_pin: !can0:E_ENABLE

# New Config Value = Old Config Value * (Actual Extruded Amount/Target Extruded Amount)
rotation_distance: 22.4748405
gear_ratio: 50:8 #50:17               #BMG Gear Ratio
microsteps: 32
full_steps_per_rotation: 200

nozzle_diameter: 0.400
filament_diameter: 1.75
max_extrude_only_distance: 110
heater_pin: can0:E_HEATER
sensor_type: ATC Semitec 104NT-4-R025H42G
sensor_pin: can0:E_TEMPERATURE
min_temp: 0
max_temp: 290
max_power: 1.0
min_extrude_temp: 172
control = pid

max_extrude_only_distance: 1400.0
max_extrude_only_velocity: 75.0
max_extrude_only_accel: 1500

pwm_cycle_time: 0.01612

pid_Kp=26.658 
pid_Ki=2.913 
pid_Kd=60.980

[tmc2209 extruder]
#uart_pin: E_TMCUART
uart_pin: can0:E_UART
interpolate: false
run_current: 0.31 # can go up to 0.35 if needed
sense_resistor: 0.110
stealthchop_threshold: 0

Surion79, 2023-12-07:
Thanks for the request, Eric. I added a PR with the changes. is the rotation_distance the default value for the sherpa mini or is it already custom tuned?

EricZimmerman, 2023-12-07:
That was my custom tuned one.