Frix-x/klippain #384: Idle timeout tweak

EricZimmerman, 2023-12-07:
### Describe the feature or hardware support you'd like

Add a note on idle timeout, so people know it happened

### Additional context or information

```
[idle_timeout]
timeout: 7200
gcode:
    RESPOND MSG="Idle timeout reached"
```

EricZimmerman, 2023-12-07:
do you prefer issues or would you rather me do PRs?

Surion79, 2023-12-07:
i do prefer issues, so it is easier to do changes for integration to Klippain. Some PRs tend to stale when changes are requested and never done.
But the quality is good. snippets of source code and some description i prefer than just descriptive text

EricZimmerman, 2023-12-07:
ok, i can do that. fast enough turn around! hehe

Surion79, 2023-12-07:
i know you can do that :) I am thinking about adding the repo saving of the manual settings you provided in this

EricZimmerman, 2023-12-07:
I bumped my timeout too so I could hear soak for an hour on the 2.4

closed 2023-12-08 by Frix-x

Benoitone, 2023-12-20:
@Surion79 @EricZimmerman do you know if there is a way to tweak timeout time without having to restart? For my exemple, I sometimes use my enclosure to dry my spool @50degres and I set timeout to 18000!!!

Surion79, 2023-12-20:
maybe with a delayed gcode command to "emulate" non idle?

EricZimmerman, 2023-12-20:
SET_IDLE_TIMEOUT i believe

Benoitone, 2023-12-20:
> SET_IDLE_TIMEOUT i believe

THANKS 🙏🏻