Frix-x/klippain #388: mainsail update manager does not see new commits in main

EricZimmerman, 2023-12-08:
### Klippain branch

- [X] I confirm using the main branch

### Version

v4.2.0-11-g822f3bc2

### Describe the bug and expected behavior

when refreshing, i would expect to see the new commits in mainsail, but i do not see anything new.

a git status from ssh shows 

```
On branch main
Your branch is behind 'origin/main' by 3 commits, and can be fast-forwarded.
  (use "git pull" to update your local branch)

nothing to commit, working tree clean
```

is it the channel: beta that causes it to not be seen? had a user today (Joel) wonder why he didnt see updates in mainsail, but when we did it via ssh, he did.

### Additional information and klippy.log

```
[update_manager Klippain]
type: git_repo
path: ~/klippain_config
channel: beta
origin: https://github.com/Frix-x/klippain.git
primary_branch: main
managed_services: moonraker klipper
install_script: install.sh
```

Benoitone, 2023-12-08:
Yes, I have the same problem with another branch and by removing Channel: beta that re-work

Surion79, 2023-12-10:
i can confirm this "unpleasant" behaviour

Frix-x, 2023-12-13:
Hey, this is indeed done on purpose:
- `channel: beta` will only sync on release tags and I think it's a good default behavior to get only the "stable" versions
- `channel: dev` will sync on every commit and will be a little bit unstable. But I tend to try to push only stable things to the main branch so it should still be good to go if you prefer fresh things

Would you prefer the dev channel by default for everyone?

EricZimmerman, 2023-12-13:
ah hah. i dont mind this then. the concern was someone wanted sherpa mini commit, and mainsail didnt show it, but git pull from the shell DID pull it.

i see the logic here as you explained and agree with it. 

i assume like anything else people could just put the klippain update check in overrides and change it to beta or whatever they want? perhaps just adding that to the docs would be sufficient

Frix-x, 2023-12-13:
> i assume like anything else people could just put the klippain update check in overrides and change it to beta or whatever they want? perhaps just adding that to the docs would be sufficient

The moonraker.conf file is yours also so you can change anything in it the same way overrides are working :)

Surion79, 2023-12-13:
Since klippain prod is stable and very good maintained qualitywise, i suggest going dev for it.

Surion79, 2023-12-13:
using beta, puts more work on you, Frix

Frix-x, 2023-12-15:
Yes I'm ok with this. Let's change it then :)

closed 2023-12-15 by Frix-x