Frix-x/klippain #393: Adding a shutdown option and adapting the case light control on the mini12864 display for Klippain.

Elias23, 2023-12-12:
### Describe the feature or hardware support you'd like

I'd love to see some options to configure the mini12864 menu via klippain. 
Below are some of the changes I added via the override file, I reckon the shutdown and the adjustment of the light control would be small quality of life improvements. I'm happy to make a PR, just not quite sure where the menu configs should go in the overall file structure.

```
#-------------------------#
#   DISPLAY                    #
#-------------------------#

[menu __main __setup __shutdown]
type: command
enable: {not printer.idle_timeout.state == "Printing"}
name: Shut down
gcode: SHUTDOWN

### menu octoprint ###
[menu __main __octoprint]
type: disabled

[menu __main __control __caselightpwm]
type: input
enable: {'output_pin caselight' in printer}
name: Lights: {'%3d' % (menu.input*100)}%
input: {printer['output_pin caselight'].value}
input_min: 0.0
input_max: 1.0
input_step: 0.01
gcode:
    SET_PIN PIN=caselight VALUE={menu.input*100}
```

Surion79, 2023-12-12:
nice idea, i think the right place would be an own cfg. but i am not sure if it should be some sort of predefined cfg (like extruder, fans etc) or a modifiable like printer.cfg, mcu.cfg, overrides.cfg.

@Frix-x what do you think?

Frix-x, 2023-12-13:
I see it a valuable to have some kind of "Klippain" controls added to the minidisplay files for everyone indeed. So this could be included automatically if an user select one of the display supporting klipper menu entries.

At least for the lights, but this should work also for the filaments changes macros or some special commands like the CLEAN/PURGE that oculd be called from it for example. What do you think about this?

PS: I do not have any display on any of my machines so I can't test it on my side

Elias23, 2023-12-13:
For the light controls they are already there by in the [default menu items](https://github.com/Klipper3d/klipper/blob/master/klippy/extras/display/menu.cfg) but assume a 0 to 1 value range instead of the 0 to 100 Klippain uses.
I haven't experimented with manual filament changes yet but. If those could be done only with the display, it would be great, maybe they already work I'd have to try it out.

Some nice things to have would be:
- Move to Park
- Activate/Deactivate Probe
- Purge
- Clean
- Heat soak with Beeper (not sure if it is really useful)
- Shutdown [copied mine from Ellis](https://ellis3dp.com/Print-Tuning-Guide/articles/useful_macros/shut_down_host.html)
- Setting Colors of RGB LEDs (for non status LEDs like case or minidisplay)

A bit more involved and probably overkill to integrate the beeper of the display into the klippain macros, [example](https://ellis3dp.com/Print-Tuning-Guide/articles/useful_macros/beeper.html)

Surion79, 2023-12-13:
when i had my printer in my office, i player thunderstruck beeper on each print end ;) :D

But that only worked because Octopus has in some versions a beeper. since i have no mini display, no beeper for me through that way.

closed 2024-02-11 by Elias23