Frix-x/klippain #394: MCU Maps!

EricZimmerman, 2023-12-13:
### Describe the feature or hardware support you'd like

For each MCU, find a nice photograph (like the one below) and diagram up the klippain PIN aliases as a visual map for what is on the board vs what klippain calls it. This will allow for people to visually see where they have connected things and then reference that in their configs. it also lets people see what klippain expects as far as which endstop is which, etc

I did two styles. one with both things (RGB) vs what the "final" alias is.

the reason for this is that i had to reference a few files when verifying my connections in order to understand the relationship between

MCU -> mcu_pin name -> klippain alias (in mcu.cfg)

### Additional context or information

[image]

Frix-x, 2023-12-13:
Yes, this is some work that I wanted to do indeed. But it's a lot of work ^^
I'll try to do it somedays but if someone wants to do a PR in the meantime, I'm really open for it 😎

EricZimmerman, 2023-12-13:
any proposed format, or is something like i have started above sufficient? id probably move the labels away more and have arrows pointing into the pins

Surion79, 2023-12-13:
Could you please give me some more info? Do you want to have a link to a customized picture in klippain for references of the hardware? Or do you want to write the mapping within klippain?
Last part is something which i took always from the sample config by the supplier.

EricZimmerman, 2023-12-13:
I want pngs that visually show the Ebb36 and how the klippain pin aliases map to what would be used in MCU.cfg

That way people can see where they have plugged stuff into the PCB and adjust the MCU.cfg file as needed so their physical connections match the aliases.

Surion79, 2023-12-13:
So for each Pin you add the klippain reference naming? could you give me an practical example for a pin in the png? In my current mindmap, it looks like a very busy png with more klippain infos than the original ones

EricZimmerman, 2023-12-13:
I already put a picture with 2 such examples here.

EricZimmerman, 2023-12-13:
Yes it would be labels for every pin but it was a pain for me to figure out which aliases actually mapped to the physical pin on my EBB when I converted to klippain

I had to flip flop some stuff around because of how I connected things like the hot end fan versus the park cooling fan and so it was a bit of a tedious process mapping between the actual pen name, the intermediate alias and what is in the MCU file

EricZimmerman, 2023-12-13:
example.

this is from my mcu.cfg

```

    E_HEATER=MCU_HOTEND0 , E_TEMPERATURE=MCU_TH0 ,

    PART_FAN=MCU_FAN2 , E_FAN=MCU_FAN1 ,
```

how do i know what those actual pins are? Here is what i have to do:

go to this file:

`config/mcu_definitions/toolhead/BTT_EBB36-42_v1.2.cfg`

to see that:

```

    MCU_HOTEND0=PB13 ,
    MCU_TH0=PA3 ,

    MCU_FAN1=PA0 , MCU_FAN2=PA1 ,
```

and now i know 

E_HEATER=MCU_HOTEND0=PB13

so rather than juggle having different config files open at at time, the png could be referenced in one of the two styles below:

[image]

in my actual config i had to flip MCU_FAN1 and MCU_FAN2 as i wired them opposite of the default MCU config in the `BTT_EBB36-42_v1.2.cfg` file

Surion79, 2023-12-14:
aaaah, i saw these texts, but i wasn't sure if that was meant. 
so it is visualizing the alias in the "read only" cfgs, correct?

EricZimmerman, 2023-12-14:
Yes so if you have to remap pins in MCU.cfg you have a simple visual aid to match up with what you see on your board in front of you.

Frix-x, 2023-12-15:
There is already this markdown documentation: https://github.com/Frix-x/klippain/blob/main/docs/pinout.md
I think this could be easily extended with link to each board doc to something more detailed with the infos you want to add Eric :)

Also FYI, I plan to add a jekyll auto-deployed GitHub page that will use these markdown files so it's definitely a good idea to put thing in there as they will be referenced in this system.

I just wonder how to do it since static images are pretty hard to maintain and would need to be fully modified and replaced again each time we do a change to the pinout in the templates... What about a doc with the default manufacturer image of the board and then some markdown table under it with the corresponding names. I even think that this table could be autogenerated by a github action script that read the mcu templates. What do you think about this?

EricZimmerman, 2023-12-15:
That would be great and serve to allow for the same thing. Visual connection of things and the corresponding alias. Love it.

Surion79, 2023-12-15:
since we already went for detailed matching on the manufacturers pinout naming, the autogeneration is realistic for the current quality of the templates.

Surion79, 2023-12-15:
If someone can give me a hint, where to start, i can put some time into it. It helps me for another business project I am working on

smwoodward, 2024-03-31:
> If someone can give me a hint, where to start, i can put some time into it. It helps me for another business project I am working on
[image]
Are you talking about something like this on a hint on where to start?

EricZimmerman, 2024-03-31:
A pic of the actual board and the board pins to intermediate and final klippain aliases.

smwoodward, 2024-03-31:
Using what RatOS uses might be a good starting point.

EricZimmerman, 2024-03-31:
Got a link? I'll look too but I'm not familiar. 

I can mock something up too.

EricZimmerman, 2024-03-31:
Looking at a few, kinda?

smwoodward, 2024-03-31:
https://github.com/Rat-OS/RatOS-configuration/tree/v2.x

Under the boards folder is where all of the different boards are, and it's the .svg files that has the wiring diagram.

closed 2025-03-22 by EricZimmerman