Frix-x/klippain #395: Issue with Flow Stability on Voron Trident When Printing Small and Complex First Layers

liustar1989, 2023-12-14:
### Klippain branch

- [X] I confirm using the main branch

### Version

v4.2.0

### Describe the bug and expected behavior

I'm encountering an issue with my Voron Trident printer, specifically when printing the first layer of small and complex parts. The problem manifests as a gradually decreasing and unstable flow rate. After troubleshooting all possible causes, I've narrowed it down to an issue with the Klipper configuration.

When I use the original configuration, the flow rate is normal (as shown in Image 1). However, when I switch to using Klippain, the flow rate becomes abnormal (as demonstrated in Image 2 3).

Below are my configuration details. In addition to these, I have also modified the /config/software/z_calibration.cfg file in Klippain and added two files: /config/mcu_definitions/toolhead/EZ36.cfg and /config/hardware/accelerometers/adxl345_eztb36.cfg. Additionally, I included 'autotune_tmc'...
[image]
[image]
[image]

### Additional information and klippy.log

[klippain.zip](https://github.com/Frix-x/klippain/files/13669412/klippain.zip)
[klippy.log](https://github.com/Frix-x/klippain/files/13669418/klippy.log)

Surion79, 2023-12-14:
is it possible to provide the full set of your mentioned "default config", the gcode file and your complete "klippain config"? the only thing i would compare would be some stuff like rotation distance or similar. if gcode is both the same, the are only little stuff that can have an impact (rotation distance, currents etc)

liustar1989, 2023-12-15:
Klippain.zip is my complete Klippain configuration, and 1.zip is the configuration I used for successful prints. The rotation_distance settings are consistent in both. The G-code file F1_2023_Kit_Card_PLA_1h2m.gcode within 1.zip was sliced using Orca, and ss_F1_2023_Kit_Card.gcode was sliced using SuperSlicer.

Image 1 shows the print result of F1_2023_Kit_Card_PLA_1h2m.gcode using the configuration from 1.zip. Image 2 displays the print using the Klippain configuration for the same G-code file. Image 3 is the result of printing ss_F1_2023_Kit_Card.gcode with the Klippain configuration.

Since this issue only occurs while printing small and complex models, with the first layer printing normally and the problem not becoming apparent until partway through the print, particularly when the outer walls start showing noticeably thinner lines, I suspect that one of the issues might be related to the Z-axis height increasing, insufficient extrusion, or a clogged nozzle.

I'm wondering if this could be due to a specific parameter causing an overload on the Z or E motors, or possibly a failure in the bed mesh, especially when printing complex models.
[klippain.zip](https://github.com/Frix-x/klippain/files/13669412/klippain.zip)
[1.zip](https://github.com/Frix-x/klippain/files/13679640/1.zip)
[image]

[image]

liustar1989, 2023-12-15:
> is it possible to provide the full set of your mentioned "default config", the gcode file and your complete "klippain config"? the only thing i would compare would be some stuff like rotation [...]

Klippain.zip is my complete Klippain configuration, and 1.zip is the configuration I used for successful prints. The rotation_distance settings are consistent in both. The G-code file F1_2023_Kit_Card_PLA_1h2m.gcode within 1.zip was sliced using Orca, and ss_F1_2023_Kit_Card.gcode was sliced using SuperSlicer.

Image 1 shows the print result of F1_2023_Kit_Card_PLA_1h2m.gcode using the configuration from 1.zip. Image 2 displays the print using the Klippain configuration for the same G-code file. Image 3 is the result of printing ss_F1_2023_Kit_Card.gcode with the Klippain configuration.

Since this issue only occurs while printing small and complex models, with the first layer printing normally and the problem not becoming apparent until partway through the print, particularly when the outer walls start showing noticeably thinner lines, I suspect that one of the issues might be related to the Z-axis height increasing, insufficient extrusion, or a clogged nozzle.

I'm wondering if this could be due to a specific parameter causing an overload on the Z or E motors, or possibly a failure in the bed mesh, especially when printing complex models.
[klippain.zip](https://github.com/Frix-x/klippain/files/13669412/klippain.zip)
[1.zip](https://github.com/Frix-x/klippain/files/13679640/1.zip)
[image]

[image]

Surion79, 2023-12-15:
wait, you used different slicers and different gcode files to compare different printer configurations?
In the name of process of elimination, please...
1. Create a single gcode file where you know that there is a different print quality when printing with different configurations
2. Use the file mentioned in 1 to print the file with your printer with your "own" configuration
3. Use the file mentioned in 1 to print the file with your printer with your "klippain" configuration.
4. Round up the single gcode file, the klippy.log of your own config renaming it to "klippy_ownConfig.log" and a klippain klippy.log renamed to "klippy_klippain.log" and zip these and attach to this issue.
5. Make pictures of your print issues and mark them

I think this is the only way to get a hold on this reported issue

closed 2024-02-01 by Frix-x