Frix-x/klippain #397: Add BTT S2DW accelerometer

EricZimmerman, 2023-12-15:
### Describe the feature or hardware support you'd like

```

[mcu btt_lis2dw]
serial: /dev/serial/by-id/usb-Klipper_rp2040_45503571290BD3C8-if00

[lis2dw]
cs_pin: btt_lis2dw:gpio9
#spi_bus: spi1a
spi_software_sclk_pin: btt_lis2dw:gpio10
spi_software_mosi_pin: btt_lis2dw:gpio11
spi_software_miso_pin: btt_lis2dw:gpio8
axes_map: -y,x,-z

[resonance_tester]
probe_points: 100, 100, 20
accel_chip: lis2dw
```

as working example

### Additional context or information

https://github.com/bigtreetech/LIS2DW

https://github.com/bigtreetech/LIS2DW/blob/master/BIGTREETECH%20S2DW%20V1.0%20User%20Manual_20231205.pdf

https://github.com/bigtreetech/LIS2DW/blob/master/Firmware/sample-bigtreetech-lis2dw-v1.0.cfg

Surion79, 2023-12-15:
As Collaborator you are free to create your own PRs ;) (When I was writing about issues, I didn't knew that Frix ranked you up :D )

EricZimmerman, 2023-12-15:
ok, ill fork and add it

Surion79, 2023-12-15:
why fork? Just add a branch :D

EricZimmerman, 2023-12-15:
yea i see how its wanting me to do it. done and the PR is almost there

closed 2023-12-15 by EricZimmerman