Frix-x/klippain #399: Shake n Tune standalone different than Klippain version

EricZimmerman, 2023-12-15:
### Klippain branch

- [X] I confirm using the main branch

### Version

latest

### Describe the bug and expected behavior

should an effort be made for them to essentially be the same, as far as the shell scripts they call, the names, etc?

i had a heck of a time installing shake n tune after klippain after klippain belt tester broke due to missing matplotlib.

EricZimmerman, 2023-12-15:
Discord convo here

https://discordapp.com/channels/460117602945990666/1096701708643614820/1185238330313547917

Surion79, 2023-12-15:
matplotlib is another issue which is already planned for next version. 
There are two ways this goes, either some sort of github action to auto-update this repo with new version of the stand alone macro or just split them apart and have both their own "history". Current situation is not ideal.

EricZimmerman, 2023-12-16:
id rather have them be 2 separate repos that have nothing to do with each other. that way you do not have to deal with messy integration. klippain installer could simply also install Shake N Tune so its "transparent" as well

Surion79, 2023-12-16:
@Frix-x this would need to be done by you since your shake and tune install script blocks install if klippain is used.
Would require a combined change in both repos.

Frix-x, 2023-12-18:
In the meantime, no need to install Shake&Tune when running Klippain since everything is already packed inside (but it's v1.x). If you try to run Shake&Tune installer, it will now work but will mess with the files so I would highly suggest to refrain doing this in order to avoid problems while I add the changes on Klippain side

roykrikke, 2024-01-31:
> This is done in the last develop commit :)

Great, looking forward for the released version.

closed 2024-02-26 by Frix-x