Frix-x/klippain #402: Auto Z calibration

tobi466, 2023-12-16:
### Klippain branch

- [X] I confirm using the main branch

### Version

V4.02

### Describe the bug and expected behavior

I Hello! I am a new user of Klipper and am absolutely thrilled :) Thank you for the good work.

I have a question.
I want to perform the auto Z calibration only once, and for the subsequent prints, I want to keep the Z offset so that it overrides the auto Z calibration when i start the Print. When I save the determined Z offset via MainSail, I get this message '' .

tobi466, 2023-12-16:
" SAVE_CONFIG section 'stepper_z' Option 'position_endstop' conflicts with included value "

EricZimmerman, 2023-12-16:
Don't use save config. You have to manually update the new offset in overrides

EricZimmerman, 2023-12-16:
However with autoz you should be adjusting switch offset. Nothing else.

Surion79, 2023-12-17:
@tobi466 have you used the manual of auto z? It seems you haven't followed it correctly. What Eric says is the essence and use case of auto z

tobi466, 2023-12-18:
Alright, I will read the instructions again and try to understand.
 I have one problem remaining. Every time I pause the print and resume, I lose my Z offset. Is this known with Klippain?

Surion79, 2023-12-19:
Changed the title to reflect the possible issue

Surion79, 2023-12-19:
Since i don't use auto-z, i can't give any info

arcadien, 2024-05-18:
Hi, i want to contribute on this one. I cannot use pause because i loose some part of the gcode, and/or the Z reference. Below are two pictures: in a paused state, and in another paused state with a clear view on lost gcode + under extrusion due to bad Z position.

First pause
[image]

Second pause
[image]

arcadien, 2024-05-18:
Little wad on the lower right corner seems to indicate this is a Z position error, and some filament has been collected on the nozzle and dropped in one time.

arcadien, 2024-05-24:
after some deeper search, the problems seems to come with the integration with Mainsail, and not directly related to Klippain.