Frix-x/klippain #409: fresh install multiple requests

Pepeuch, 2023-12-19:
### Describe the feature or hardware support you'd like

hello, I am currently installing klippain for the first time, I find the idea excellent to be able to facilitate and standardize the configurations and thus create a standard.
However, I have noticed several gaps and certain inconsistencies which I have identified and which I am submitting to you in order to improve compatibility a little further.

first of all the addition of TMC5160 for Z axes, which has not yet been implemented in your configuration.

then why use the HE1 PIN on a keenovo heated bed when the BED0 PIN would be more logical (observed on an octopus pro).

there is also the SPI bus of the mini12864 screens which is shared with that of the TMC drivers and Klipper does not like that at all, this creates an error, I got around the problem by replacing the SPI bus of the screen with the one which is reserved for the 8266 wifi module (SPI2).
because this bus is not used by klipper since everything goes through the raspberry. here are the PINS used to replace for this to work (this of course requires a modification of the screen cables):
EXP1_3=PB12
EXP2_1=PC2
EXP2_2=PB13
EXP2_6=PC3

thanking you for the work done for this configuration.

Surion79, 2023-12-19:
you can always extend klippain for such things. Klippain is for every experience level and tmc_autotune is not for basic users. especially if they use a fully prepared kit, there is no need to burden them with something they don't require.
Regarding the other points:
- TMC5160. no feature request for them yet. (you can open a PR if you need it)
- Pins setting: some manufacturers are multiplexing pins. Klippain can't detect, what is best for you. other people use wifi and no display. Thats why Klippain has an open approach regarding configuration. It cannot be done fully automated.

Pepeuch, 2023-12-19:
Thank you for your response, I will take note and I will make a PR for the 5160.

Regarding the pinout for the screens you should test yourself as this creates an error, and the aim is precisely to avoid this and doubt beginners. the fact is that standardizing configurations also allows you to discover bugs. maybe you should add the fact that you have to ask; after selecting the drivers if a screen is connected in which case you will need to ask which SPI port used. by indicating to the user that he cannot use the same SPI port as those of the drivers.

And when you tell me that users use wifi, yes they use it but only that of the rpi, I'm talking about that of the motherboard. klipper probably does not communicate via wifi with the card. it is therefore a more than viable option to use SPI2.

Frix-x, 2023-12-19:
Thanks for your PR. I also saw your message on the French Discord :)

So basically I would be really happy to merge your changes with the TMC5160 to make it available for everyone.
Regarding the SPI topic, it's not so easy because usually on the manufacturer documentation the screen is attached to one of the SPI bus and it's the one we try to use in Klippain to avoid confusion. For your specific MCU it is indeed complicated and I don't have an easy solution. The easy solution I can provide quickly is a comment in the template for example?

Regarding the TMC autotune package, I'm actually one of the main authors of this plugin with andrew (and we're working on improving it with a new feature that will come in the near future). But since it's a pretty standalone plugin, nothing needs to be added in Klippain for it: just install it and add the [autotune_tmc] sections to your overrides.cfg or mcu.cfg as you prefer.

closed 2024-02-26 by Frix-x