Frix-x/klippain #413: Support Mellow Fly SB2040 Pro

flyespresso, 2023-12-21:
### Describe the feature or hardware support you'd like

Is it possible to get this toolhead MCU supported by Klippain?

All the details are available (Upper right you can change to english/other language):
https://mellow.klipper.cn/#/board/fly_sb2040_pro/README

Pinouts are available on the sidebar too.

Not sure if there is other info that is needed? 

### Additional context or information

Main page: https://mellow.klipper.cn/#/board/fly_sb2040_pro/README
Pinouts: https://mellow.klipper.cn/#/board/fly_sb2040_pro/pins
Reference Config: https://mellow.klipper.cn/#/board/fly_sb2040_pro/cfg
Wiring Info: https://mellow.klipper.cn/#/board/fly_sb2040_pro/wiring

Benoitone, 2023-12-21:
I can make this today...

Benoitone, 2023-12-21:
#417 I just create a PR for Mellow SB2040 v2 and SB2040 Pro toolhead boards based on your infos. Can you have a check on it?

flyespresso, 2023-12-21:
OMG amazing!

Yeah I double checked the pins and I think what you have is spot on. Thank you so much!

Frix-x, 2024-02-01:
Closing this issue as it was merged in the develop branch and will be released very soon in the main branch :)

closed 2024-02-01 by Frix-x