Frix-x/klippain #414: Support for BD Sensor

Kiiiddd, 2023-12-21:
### Describe the feature or hardware support you'd like

With multiple Brands coming out with a repackaged PandaPi BD Sensor(BTT is the latest one with the EDDY sensor) it would would probably be good to have it as a option for a sensor.

The GitHub for that sensor is 
https://github.com/markniu/Bed_Distance_sensor

Frix-x, 2023-12-21:
Thanks for the PR.
I know some people already running this sensor with Klippain using their overrides so it should definitely be possible

Surion79, 2023-12-21:
@Kiiiddd what is the name of the BTT product with the eddy sensor? Thanks :)

Kiiiddd, 2023-12-21:
It is called EDDY should be put in the next few days

https://twitter.com/BigTreeTech/status/1734148535371845917

f1cs2000, 2024-01-22:
@Kiiiddd Hello, I am curious if you had the BDsensor working with Klippain?
I use the BDsensor and I am thinking going with Klippain. Should I go there or not ?
Is your BDsensor working in real ttime adjust or just meshing ?

Darkjack69, 2025-01-07:
Hello,
Please integrate the sensor.

As the 2nd generation of the sensor is now available
and it is also a cost-effective alternative to the other BD sensors. It would make sense.

I have an EDDY Douo here myself but am hesitant about installing it as it has not yet arrived in the Klippein universe :(

jace, 2025-02-10:
Does anyone have working config for Eddy probe while retaining Voron Tap for Z-offset in Klippain?

zarpyj, 2025-09-19:
Please add this sensor in the list of supported sensor, I try myself to use it by addind what sould be added in overrides.cfg, but for now I dont succed in this approch
Thank you in advance