Frix-x/klippain #415: hotend over 150 degres before bedmesh in print start

Pepeuch, 2023-12-21:
### Klippain branch

- [X] I confirm using the main branch

### Version

v4.2.0-16-ga207639

### Describe the bug and expected behavior

hello, I noticed in your print start that the temperature of the nozzle is higher than 150°C during the meshing of the bed, is it possible to modify the print start so that the meshing is done with an adequate temperature?

thank you again for your work.

Frix-x, 2023-12-21:
What kind of probe are you using?

If using TAP it should not be the case as there is some safety included in Klippain and this would be a bug.
If using another kind of probe this behavior is normal to measure the bed at the printing temperature to be as close as possible as the printing conditions.

Surion79, 2023-12-21:
I use tap and haven't had the issue with an overheated nozzle.

EricZimmerman, 2023-12-21:
It's all managed appropriately depending on the configuration selected

Pepeuch, 2023-12-22:
hello, so I have a classic inductive probe, and I think it doesn't like the hotend to be too hot, and I think that this generates measurement errors, that's why I was wondering if it was normal for the temperature to increase just before the bed mesh. while for the QGL the temperature is indeed at 150. Perhaps add a temperature for the mesh not only the TAP.

Generalizing the safety temperature to 150 for the QGL/bedmesh would surely be simpler, regardless of the probe used. No?

Surion79, 2023-12-22:
Could you please describe in which case it is over 150 degrees? (I mean steps done to have it over 150)

Pepeuch, 2023-12-22:
So it's over 150 degrees when he meshes the bed. while it should be at a value of 150 (I specify that I use the original macros from klippains without any modification); and that I test all the macros to see if there are any inconsistencies/errors before launching large prints.

Surion79, 2023-12-23:
Could you add a klippy log until the start with the mesh? Can you also give an example gcode file which includes the 'misbehaviour'?
(E.g. a voron cube at 300% size to habe a bed mesh call)

Surion79, 2024-01-01:
hmm, reading your last comment again: You do a primeline and after that you do a bed mesh?

Pepeuch, 2024-01-03:
hello and happy new year to the whole team;
excuse my late response, the problem (it is not necessarily one), but the order of actions which is faulty, once the QGL is made at the correct temperature (150°C), the nozzle starts to heat to the printing temperature (200°C), then purge, clean the nozzle, do a homing, until then OK everything is good.

From this point on, the macro should lower the nozzle temperature to 150°C before meshing the bed (which is generally the backup temperature for the probes). But since this step is missing, the mesh of the bed is done at printing temperature (I specify that I am talking about the temperature of the nozzle).

I therefore ask if it was possible to add this instruction in the startup macro and even add a user variable for the temperature of the nozzle to carry out the meshing of the bed and this whatever the probe used.

Surion79, 2024-01-03:
your order is wrong. mine does primeline right before the print start. QGL, Bedmesh comes always before that.  
can you please attach a klippy.log?

Pepeuch, 2024-01-03:
Mine is:
HEATSOAK_BED
HEATSOAK_CHAMBER
TILTING
EXTRUDER_HEATING
PURGE
CLEAN
Z_CALIB
BED_MESH
PRIMELINE
EXTRUDER_PREHEATING

and the temperature of the hotend does not go down before the BED_MESH

EricZimmerman, 2024-01-03:
Why not swap preheat and heating? Meshing at 150 is fine and then it doesn't need to drop.

EricZimmerman, 2024-01-04:
I've not seen a Need to purge with klippain since it does retractions at the end of a print, personally

Surion79, 2024-01-04:
i think the printstart_actions must be different. try 

HEATSOAK_BED
HEATSOAK_CHAMBER
TILTING
EXTRUDER_HEATING
PURGE
CLEAN
EXTRUDER_PREHEATING <<<----
Z_CALIB
BED_MESH
PRIMELINE
EXTRUDER_HEATING <<<---

preheating issues a M109 150

Surion79, 2024-01-04:
hmm, i understand if someone with a Voron Taps wants to purge/clean nozzle before a bed mesh, But I don't see the reason to do it on an inductive probe.

I have Voron Tap and don't do a prior purge/clean as it is the same for me as for Eric. I have perfect first layer

Surion79, 2024-01-04:
please change order to

HEATSOAK_BED
HEATSOAK_CHAMBER
TILTING
EXTRUDER_HEATING
PURGE
CLEAN
EXTRUDER_PREHEATING <<<<---
Z_CALIB
BED_MESH
EXTRUDER_HEATING <<<-
PRIMELINE

PREHEATING issues a M109 S150 which is then waiting to reach 150 degrees up from the previous heated value

Surion79, 2024-01-04:
putting primeline before extruder_heating won't work as the function does not have any temperature commands afais

Surion79, 2024-01-04:
you added no klippy log, just a link to this issue.

if the print_start action is really as you describe, the print can never begin, since the extruder is at safe temperature, not at printing temperature.
if you manipulated the value for variable `safe_extruder_temp` to the printing temperature then your request will never work