Frix-x/klippain #421: Extruder add printing temperature but Klippain still goes to save temperature

roykrikke, 2023-12-23:
### Describe the feature or hardware support you'd like

If I preheat my extruder. Klippain is not checking if the extruder is already add the needed printing temperature. I don't know if there's a reason for that. I think a check can be built in to ensure that the temperature does not first return to extruder safe temperature aka variable_safe_extruder_temp from variables.cfg.

Surion79, 2023-12-23:
Can you explain a bit more? Sounds like an issue to check.
Klippy log until error (extruder not at temp) would be good atarting point

roykrikke, 2023-12-23:
I think this image shows what I explained. Around 15:36 I started the print. Around 15:30 I started the preheat, but manual.
[image]

Surion79, 2023-12-23:
why do you preheat your extruder?

roykrikke, 2023-12-23:
I just this time preheat manual for no good reason. I just noticed that "klippain" first went back to safe temp on the extruder. I don't think that's necessary. Klippain makes the assumption that the extruder is cold. Which might also make sense, but if it's already on temperature. My thought was: why cool down first.

EricZimmerman, 2023-12-23:
You need to if using tap but not necessarily for other probes

roykrikke, 2023-12-23:
> You need to if using tap but not necessarily for other probes

That not the point. The point is just check if extruder is hot and don't cool down if not needed. Under any condition.

EricZimmerman, 2023-12-23:
It is the point. Tap has to be taken into consideration else it probes the bed too hot. 

For non tap sure those checks can be added but that is hardly under any condition

roykrikke, 2023-12-23:
> It is the point. Tap has to be taken into consideration else it probes the bed too hot.
> 
> For non tap sure those checks can be added but that is hardly under any condition

Oke now I get your point. I not using a tab and therefor did not think about this.

Surion79, 2024-01-01:
i close the issue since it is cleard in my understanding. Nothing to be done

closed 2024-01-01 by Surion79