Frix-x/klippain #423: option to keep probe attached after leveling/probing/meshing.

teookie, 2023-12-30:
### Describe the feature or hardware support you'd like

Would be nice to keep the probe attached between consecutive `START_PRINT` modules that both utilize it. For example, keep the probe attached after QGL because the very next task is BED_MESH.

I've done this in the past (without Klippain) by passing a KEEP_PROBE parameter to each task. To achieve the same functionality in Klippain I would have to override several macros and modules, which is not ideal.

Surion79, 2024-01-01:
Klippain uses a free configurable sequence (via modules) for start print. how do you think we can add this feature to it? just passing keep_probe would not be enough possibly.

teookie, 2024-01-01:
Maybe all the modules should pass {rawparams} to each other, or, there could be specific modules for attaching and docking the probe. I dunno, just spit balling.

Surion79, 2024-01-01:
the modules are designed to be independant to a specific degree. otherwise, the sequence would not be customizable. rawparams is not working, if you do not use the params to do something. you would need to override macros anyway for that.

i think your "solution" would be to use a custom module where you combine your probe related activities and add it in the start sequence, AND removing the modules which you replaced, so they are not executed twice.

Frix-x, 2024-01-10:
@Surion79 I think what teookie want is not to configure the sequence differently but to avoid docking the probe if the next module is using it in order to avoid a direct docking and undocking.

@teookie, you are right, it was like this in the past before I split the START_PRINT into submodules. But the thing is that now that the sequence is configurable, I don't know which module is called when and it make it more difficult to predict it. But this is something I want to reintroduce as well since it also bother me 😆 I just need to find a smart way to do this prediction at the beginning of the START_PRINT sequence, maybe with a pre-analysis of the sequence, and then pass it as a parameter to each macro to tell them if they need to dock or not.

EricZimmerman, 2024-01-10:
could you introduce another macro that locks things and put it in the order for print start? same with unlocking.

it would be more calls, but if its just lock and unlock, it shouldn't be noticeable

Surion79, 2024-01-10:
@Frix-x I understood that correctly. I was pointing out that there is no direct connectivity possible, since the modules can be ordered in a generic way. 
If we introduce any kind of "lock" mechanism, it would require to modify most modules, since there needs to be always a check if a dock is not "allowed". Currently the modules are encapsulated, so they only use probe if required. If we had some "locking" functionality, it would require modification in all modules, since it is now possible to have the probe attached after completion of each module. and the finished module does not know if the next "inline" module requires the dock. So the check needs to be implemented in every module. Even in the custom ones. (if attached and setting "keep probe attached" == true but only for modules which are compatible with a docked probe, so EXTRUDER_HEATING would require a hard "dock probe" call if probe is attached )
And using separated "lock"/"unlock" modules will also not work effectively. I tried implementing an additional set for printstart_actions in combination with HH but it won't work. Combination of HH with any other configuration within printstart_actions doesn't work. Therefore the "solution" for HH is to put the macros "hardwired" at specific locations within the start_print macro.

My proposed solution for teookie was to copy the modules in a combined custom module to leave the dock out in between.