Frix-x/klippain #440: Octopus Pro - HE0 and Driver 3 Enable Swapped

awickham10, 2024-01-23:
### Klippain branch

- [X] I confirm using the main branch

### Version

4.2.0

### Describe the bug and expected behavior

The board pinout for the Octopus Pro 1.1 has the pins swapped for HE0 and Driver 3 Enable. I've confirmed this with my old configuration and the BTT documentation.

/config/mcu_definitions/main/BTT_Octopus_Pro_v1.1.cfg

MCU_MOTOR3_ENABLE=PA2 should be MCU_MOTOR3_ENABLE=PA0
MCU_HE0=PA0 should be MCU_HE0=PA2

Supporting documentation:
https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-Pro/blob/master/Hardware/BIGTREETECH-Octopus-Pro-V1.0-Color-PIN-V3.0.pdf

Benoitone, 2024-01-23:
Hello, I think you have the octopus pro V1.0 or 1.0.1, and the pinout link is for V1.0 and not for V1.1 has the Klippain config file is for.
They're some differences between 1.0 and 1.1: [have a look here](https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-Pro/tree/master#changes-in-v11-compared-to-v10)
I don't have octopus board, but I think if you have an octopus pro 1.0 you must use the BTT_Octopus.cfg klippain board file.
The BTT_Octopus_Pro_v1.1.cfg is for v1.1

awickham10, 2024-01-23:
I think you're correct! That summary of changes looks exactly like what I've had to modify. I'll switch over that config. Sorry for wasting your time!

Is it worth adding an Octopus Pro 1.0 config file to avoid confusion? I personally didn't realize I needed to use the Octopus config. I'd be happy to submit a PR for it if you think it's worth it.

Benoitone, 2024-01-23:
I found it's a little bit confused to use the octopus template for octopus pro v1.0 but maybe duplicate the template will be easier. But maybe @Frix-x can answer for that?

Frix-x, 2024-01-25:
Hi, I agree, this can be quite confusing.

Since it's a user file, I don't see any other choice but to duplicate it. However, the line referencing the manufacturer board pin can stay on the original octopus (that way we don't duplicate the manufacturer board pin as well).

awickham10, 2024-01-25:
For sure. I'll get one submitted today!

awickham10, 2024-01-25:
I created a PR with the change. I believe the direction was to create a BTT Octopus Pro v1.0 user template. Let me know if I understood that wrong (I'm pretty new to the framework).

closed 2024-01-29 by Frix-x