Frix-x/klippain #447: Skew correction from Klipper implemented like

roykrikke, 2024-01-31:
### Describe the feature or hardware support you'd like

Would be a great feature to add support for Skew correction inside klippain. E.g part of the macro START_PRINT.
https://www.klipper3d.org/Skew_Correction.html

To enable skew correction in the printer.cfg add [skew_correction]
It is needed to enter your skew numbers into the console and save it. This will save your skew profile to the bottom of printer.cfg.
E.g. SET_SKEW XY=141.22,141.15,100 XZ=140.87,141.34,99.84 YZ=141.12,141.26,99.94
SKEW_PROFILE SAVE=my_skew
Klipper docs warns about skew correction causing problems: https://www.klipper3d.org/Skew_Correction.html#caveats

I think it would be good to add SKEW_PROFILE LOAD to your START_PRINT macro and SET_SKEW CLEAR=1 to the END_PRINT macro. This could either be add to the klippain macros for Klipper or in the slicer program. Implementing this in klippain would be great. and a nice future in my opinion. 

I'm curious what others think of this.

Surion79, 2024-01-31:
the caveats i don't like that we can't control if the head is running outside the configured values as described.
how to handle skew profiles which are not set up? it feels like we being to debug other ppls skew settings, because they are not working as they would expect it. (wrong profile, profile not set correctly)
integration is no problem for someone who requires it. you can easily override a custom start_print module. there are three of them available for additional start_print actions.

Zanonia, 2024-06-12:
I don't understand how to use the custom start print module.
I have had this code in my overrides.cfg:

```
[gcode_macro _MODULE_CUSTOM1]
gcode:
    # Load skew profile
    SKEW_PROFILE LOAD=my_skew_profile 
```

But it doesn't seem to load something at print start...