Frix-x/klippain #457: BTT Kraken support

DaWingla, 2024-02-07:
### Describe the feature or hardware support you'd like

is the support of the BTT Kraken planed??

Phantastic1, 2024-02-14:
following... I'm currently running a BTT Kraken for my MCU and would like to see a configuration for this MCU

Benoitone, 2024-02-14:
> following... I'm currently running a BTT Kraken for my MCU and would like to see a configuration for this MCU

On which printer?

DaWingla, 2024-02-14:
I run it in a voron 2.4

DaWingla, 2024-02-14:
S1 and s2 are a and b and s5 6 7 and 8 are there z drives. Like the ldo 2.4 config.

DaWingla, 2024-02-14:
I made an rough printer.cfg so that the printer works.larer I could post it

Benoitone, 2024-02-14:
ok...
can you check #476 ?

DaWingla, 2024-02-14:
Ohh cool thank you .

Frix-x, 2024-02-21:
PR merged in develop. It will be available on the next release :)

closed 2024-02-21 by Frix-x