Frix-x/klippain #459: Support for Canbus on

jrsphoto, 2024-02-08:
### Describe the feature or hardware support you'd like

On my Voron 2.4R2, I'm using an Octopus board and canbus to the [BTT EBB SB2209 or SB2240](https://biqu.equipment/products/bigtreetech-ebb-sb2209-can-v1-0) stealthburner canbus board.  Canbus can be difficult to setup, especally the first time, and having something like Klippain could help.

Surion79, 2024-02-09:
i have the exact setup with Klippain. I use a Octopus board with SB2240.
What is your issue? We have even all template files available

Surion79, 2024-02-09:
the only thing klippain cannot provide is getting the canbus uid, or serial connection string. those things are required to be provided since klippain's focus is configuration of a printer, not installing/updating firmware

jrsphoto, 2024-02-09:
Sorry about that.   I looked and It didn’t look like support was there for it.    I see it now.    Thank you.    Great project by the way.

closed 2024-02-09 by jrsphoto