Frix-x/klippain #465: Problem with PRIMELINE macro w/ stock CR-10

arcadien, 2024-02-10:
### Klippain branch

- [X] I confirm using the main branch

### Version

v4.2.0-21-gdf0aee04

### Describe the bug and expected behavior

When using PRIMELINE macro, i always get "[..]is too high and will exceed the max_extrude_cross_section!" error. As setting max_extrude_cross_section to 50 without understanding was not an option, I did some maths in a [Google Sheet](https://docs.google.com/spreadsheets/d/1fKTwtf8pzlli5XlOhl5m0oC_1Djb6MKVpYVfE9rR_NQ/edit?usp=sharing) based on code and various documentation. I found that `line_height` should be lowered to 0.5mm in my case.
I suggest to make this (currently hardcoded) as a `variables.cfg` entry.

Surion79, 2024-02-11:
https://www.klipper3d.org/Config_Reference.html#common-extruder-and-heated-bed-support

`#max_extrude_cross_section:`
`#   Maximum area (in mm^2) of an extrusion cross section (eg,`
`#   extrusion width multiplied by layer height). This setting prevents`
`#   excessive amounts of extrusion during relatively small XY moves.`
`#   If a move requests an extrusion rate that would exceed this value`
`#   it will cause an error to be returned. The default is: 4.0 *`
`#   nozzle_diameter^2`

arcadien, 2024-02-11:
yes I saw that, and a lot of internet posts as well. I went a step beyond in understanding (i hope). 
I can set 50 as max_extrude_cross_section. But that make no sense, right ? :)

Frix-x, 2024-02-13:
Yes, but this option is like a "false" safety check that doesn't add anything. Most Klipper users just set it to a higher value to avoid any problems, especially when purging filament, which is almost impossible to do with a standard value, since the role of purging is actually to purge a large amount of filament to create a blob that will attract everything and allow an easier cleaning process of the nozzle.

As a side note, there is some discussion about removing this safety check from Klipper altogether. Not sure where this discussion is now, but be aware that it could be removed in the future and be a "no problem".
Note 2: If you're using an ERCF or any MMU, it's actually mandatory to have it set to a higher value or you won't be able to do the load/unload filament moves.

Since Klippain is compatible with the ERCF/MMU, there is already some correct values set for all extruder types and this will need to stay, so I'll unlikely merge this change.

commit e0a170e0 (2024-02-20): fix: make prime line height a variable parameter (#466)

Fix #465
closed 2024-02-13 by Frix-x